dlundqvist/xone #8: Medusalix appears to be back... Merge time?

pallaswept, 2024-02-03:
Just noticed lots of commits over there in the past month. He's missing a bunch of your stuff and you're missing a bunch of his. I wonder if a merge could be arranged?

pallaswept, 2024-02-03:
Tried to suggest something over there but his issues section has been removed and PRs are locked to collaborators..... I hope you can find some way to communicate.

Al-Infinitum, 2024-02-15:
> I hope you can find some way to communicate.

I just discovered there's an active conversation with Medusalix on discord, message below specifically mentions merging the fork.
https://discord.com/channels/733964971842732042/819644054563192846/1192972953827164261

pallaswept, 2024-02-16:
> > I hope you can find some way to communicate.
> 
> I just discovered there's an active conversation with Medusalix on discord, message below specifically mentions merging the fork. https://discord.com/channels/733964971842732042/819644054563192846/1192972953827164261

Thanks for the update. I joined real quick to take a look and it's clear from several conversations that medusalix is well aware of the fork and that there are various patches that need to be merged, so hopefully that'll work itself out in time.

Fxzzi, 2024-04-05:
I think getting these merged will be very useful considering people (including myself) experience issues on base xone which this fork explicitly fixed. I get random disconnects on xone-dkms-git which I don't get here, etc

dlundqvist, 2024-05-25:
> I think getting these merged will be very useful considering people (including myself) experience issues on base xone which this fork explicitly fixed. I get random disconnects on xone-dkms-git which I don't get here, etc

Did that happen in the past to with upstream or is that a new problem?

dlundqvist, 2024-05-25:
> Just noticed lots of commits over there in the past month. He's missing a bunch of your stuff and you're missing a bunch of his. I wonder if a merge could be arranged?

Of course my life turned busy when he came back. I'll look into what he's missing from here and try to coordinate something.

Fxzzi, 2024-05-25:
> > I think getting these merged will be very useful considering people (including myself) experience issues on base xone which this fork explicitly fixed. I get random disconnects on xone-dkms-git which I don't get here, etc
> 
> Did that happen in the past to with upstream or is that a new problem?

I think it's an issue related to which band the receiver is being connected by. Not sure exactly when it started but I don't think it was any time recently

dlundqvist, 2024-05-25:
> > Did that happen in the past to with upstream or is that a new problem?
> 
> I think it's an issue related to which band the receiver is being connected by. Not sure exactly when it started but I don't think it was any time recently

I just merged in upstream's changes. I did not see any changes to channel management so it should still work for you.

dlundqvist, 2024-05-25:
I also started to create PR:s for commits upstream is missing.

pallaswept, 2024-05-25:
You've been very busy today, we all appreciate it mate!

Fxzzi, 2024-05-25:
I've created an AUR package to install this fork until everything gets merged: https://aur.archlinux.org/packages/xone-dlundqvist-dkms-git

Thanks so much for your work!!

adamdmoss, 2024-09-06:
Just popping in here to say I only just discovered this fork - and it seems to have a bunch of nice-looking stuff that upstream doesn't have yet (or have PRs for?) that still seems to leave this fork more reliable than upstream.  Kudos!
As a spectator I'm particularly intrigued by a3d95ec7f0beca0befa978c873103657e26b27ec and b3b1a37b5be216332a5b2e4e8d85665675ca486f which I suspect are solving various reliability issues for me.

dlundqvist, 2024-09-08:
Happy to help. According to author a3d95ec7f0beca0befa978c873103657e26b27ec is not the best solution, but it does seem to solve issues for many so for now I'm leaving it in. I actually just now saw a bug in b3b1a37b5be216332a5b2e4e8d85665675ca486f, where it does the wrong thing if `xone_dongle_toggle_pairing` fails. I should be able to fix that soonish.

markmb88, 2024-09-18:
6.11 brought me here. Thank you!

closed 2025-06-24 by Lawstorant