dlundqvist/xone #10: Unable to compile on Ubuntu 24.04

teeedubb, 2024-05-08:
Hi,

I am unable to compile to xone driver after updating to Ubuntu 24.04 with the following error:

`Installing xone v0.3-21-gc654623...
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der
Creating symlink /var/lib/dkms/xone/v0.3-21-gc654623/source -> /usr/src/xone-v0.3-21-gc654623

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.8.0-31-generic -C /lib/modules/6.8.0-31-generic/build M=/var/lib/dkms/xone/v0.3-21-gc654623/build...(bad exit status: 2)
ERROR (dkms apport): binary package for xone: v0.3-21-gc654623 not found
Error! Bad return status for module build on kernel: 6.8.0-31-generic (x86_64)
Consult /var/lib/dkms/xone/v0.3-21-gc654623/build/make.log for more information.
DKMS make.log for xone-v0.3-21-gc654623 for kernel 6.8.0-31-generic (x86_64)
Wed 08 May 2024 20:40:59 AEST
make: Entering directory '/usr/src/linux-headers-6.8.0-31-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  CC [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/transport/wired.o
  CC [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/transport/dongle.o
  CC [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/transport/mt76.o
  CC [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/bus/bus.o
  CC [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/bus/protocol.o
  CC [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/driver/common.o
  CC [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/driver/gamepad.o
  CC [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/driver/headset.o
  CC [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/driver/chatpad.o
  CC [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/driver/guitar.o
  LD [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/xone-gip-guitar.o
  LD [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/xone-gip-gamepad.o
  LD [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/xone-wired.o
  LD [M]  /var/lib/dkms/xone/v0.3-21-gc654623/build/xone-gip-headset.o
/var/lib/dkms/xone/v0.3-21-gc654623/build/transport/mt76.c:188:6: warning: no previous prototype for ‘xone_mt76_prep_message’ [-Wmissing-prototypes]
  188 | void xone_mt76_prep_message(struct sk_buff *skb, u32 info)
      |      ^~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/xone/v0.3-21-gc654623/build/transport/dongle.c:1026:10: error: ‘struct usb_driver’ has no member named ‘drvwrap’
 1026 |         .drvwrap.driver.shutdown = xone_dongle_shutdown,
      |          ^~~~~~~
/var/lib/dkms/xone/v0.3-21-gc654623/build/transport/dongle.c:1026:36: error: initialization of ‘const struct attribute_group **’ from incompatible pointer type ‘void (*)(struct device *)’ [-Werror=incompatible-pointer-types]
 1026 |         .drvwrap.driver.shutdown = xone_dongle_shutdown,
      |                                    ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/xone/v0.3-21-gc654623/build/transport/dongle.c:1026:36: note: (near initialization for ‘xone_dongle_driver.dev_groups’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/xone/v0.3-21-gc654623/build/transport/dongle.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.8.0-31-generic/Makefile:1926: /var/lib/dkms/xone/v0.3-21-gc654623/build] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.8.0-31-generic'
`

I am not sure what to make of it or what it means - Does anyone have any suggestions?

Thanks

dlundqvist, 2024-05-25:
Sorry for the late response. It looks like an incompatibilty with newer kernels version. Did you get this solved somehow?

dlundqvist, 2024-05-25:
This should be fixed in master now. Please let me know how it goes.

teeedubb, 2024-05-25:
Thank you, I will test and report back

dlundqvist, 2024-06-05:
Did it help?

teeedubb, 2024-06-05:
Apologies for the delayed reply - yes it compiles cleanly now on Ubuntu 24.04
  

  
Thank you

closed 2024-06-05 by dlundqvist