dlundqvist/xone #12: DKMS cannot install on 6.11 release candidate kernel

wantija, 2024-08-13:
The module compiles fine (and in a clean chroot too on Arch), but dkms cannot install it on 6.11-rc kernels, works on 6.10. 

```DKMS make.log for xone-0.3.r50.g58004bf for kernel 6.11.0-rc3-2-cachyos-rc (x86_64)
Tue 13 Aug 2024 02:29:22 AM EDT
make: Entering directory '/usr/lib/modules/6.11.0-rc3-2-cachyos-rc/build'
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/transport/wired.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/transport/dongle.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/transport/mt76.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/bus/bus.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/bus/protocol.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/auth/auth.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/auth/crypto.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/driver/common.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/driver/gamepad.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/driver/headset.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/driver/chatpad.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/driver/madcatz_strat.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/driver/madcatz_glam.o
  CC [M]  /var/lib/dkms/xone/0.3.r50.g58004bf/build/driver/pdp_jaguar.o
/var/lib/dkms/xone/0.3.r50.g58004bf/build/bus/bus.c:125:18: error: initialization of ‘int (*)(struct device *, const struct device_driver *)’ from incompatible pointer type ‘int (*)(struct device *, struct device_driver *)’ [-Wincompatible-pointer-types]
  125 |         .match = gip_bus_match,
      |                  ^~~~~~~~~~~~~
/var/lib/dkms/xone/0.3.r50.g58004bf/build/bus/bus.c:125:18: note: (near initialization for ‘gip_bus_type.match’)
make[2]: *** [scripts/Makefile.build:244: /var/lib/dkms/xone/0.3.r50.g58004bf/build/bus/bus.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/lib/modules/6.11.0-rc3-2-cachyos-rc/build/Makefile:1928: /var/lib/dkms/xone/0.3.r50.g58004bf/build] Error 2
make: *** [Makefile:224: __sub-make] Error 2
make: Leaving directory '/usr/lib/modules/6.11.0-rc3-2-cachyos-rc/build'
```

wantija, 2024-08-18:
Nevermind, [this](https://github.com/medusalix/xone/pull/48) fixes it.

closed 2024-08-18 by wantija

dlundqvist, 2024-09-08:
Sorry for being slow, above PR is in my fork now.

pallaswept, 2024-09-29:
> Sorry for being slow, above PR is in my fork now.

Thank you for maintaining the driver for us!