dlundqvist/xone #13: "Error! Installation aborted" on Manjaro

mben42, 2024-09-07:
Hi,

Hi get this error when trying to install (during install.sh run) on Manjaro. I haven't find any log or anything to understand what fails. If i retry to start install.sh, it tells me the driver is already installed, but xone-get-firmware.sh doesn't exist… I tried to uninstall and reinstall, same error.

Thanks for your help, here is the terminal output:

```zsh
[17:37:32] [~/Téléchargements/xone] [master]
-> % sudo ./install.sh
Installing xone v0.3-88-gf6f30b7...
Sign command: /usr/lib/modules/6.11.0-rc4-7-MANJARO/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/xone/v0.3-88-gf6f30b7/source -> /usr/src/xone-v0.3-88-gf6f30b7

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.11.0-rc4-7-MANJARO -C /usr/lib/modules/6.11.0-rc4-7-MANJARO/build M=/var/lib/dkms/xone/v0.3-88-gf6f30b7/build...
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-wired.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-dongle.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip-gamepad.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip-headset.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip-chatpad.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip-madcatz-strat.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip-madcatz-glam.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip-pdp-jaguar.ko
Cleaning build area...

xone-wired.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-wired.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-dongle.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-dongle.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip-gamepad.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip-gamepad.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip-headset.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip-headset.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip-chatpad.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip-chatpad.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip-madcatz-strat.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip-madcatz-strat.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip-madcatz-glam.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip-madcatz-glam.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip-pdp-jaguar.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip-pdp-jaguar.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.
Error! Installation aborted.
[17:38:43] [~/Téléchargements/xone] [master]
-> % sudo ./install.sh          
Driver is already installed!
[17:38:56] [~/Téléchargements/xone] [master]
-> % sudo xone-get-firmware.sh 
sudo: xone-get-firmware.sh : command not found
[17:39:56] [~/Téléchargements/xone] [master]
-> % locate xone-get-firmware.sh
[17:40:56] [~/Téléchargements/xone] [master]
-> % 
```

dlundqvist, 2024-09-08:
You should be able to run `sudo ./uninstall.sh` and start with `sudo ./install.sh` again. The script `xone-get-firmware.sh` is installed as one of the last steps of `install.sh`, but after DKMS (which failed for you). If you still get errors for `install.sh` please run `sudo sh -x ./install.sh` and upload the output here.

dlundqvist, 2024-09-08:
It still thinks the module is already installed, that is why `dkms` fails. What does `sudo sh -x ./uninstall.sh` say?

mben42, 2024-09-08:
> It still thinks the module is already installed, that is why `dkms` fails.

Might be possible, I installed tskaar's fork (see https://github.com/medusalix/xone/pull/48#issuecomment-2311206195), maybe it didn't removed fully.

> What does `sudo sh -x ./uninstall.sh` say?

```
[15:00:34] [~/Téléchargements/xone] [master]
-> % sudo sh -x ./uninstall.sh
[sudo] Mot de passe de mathis : 
+ set -eu
++ id -u
+ '[' 0 -ne 0 ']'
++ lsmod
++ grep '^xone_'
++ cut -d ' ' -f 1
++ tr '\n' ' '
+ modules=
++ dkms status xone
++ head -n 1
++ tr -s ,:/ ' '
++ cut -d ' ' -f 2
+ version=v0.3-88-gf6f30b7
+ '[' -n '' ']'
+ '[' -n v0.3-88-gf6f30b7 ']'
+ echo 'Uninstalling xone v0.3-88-gf6f30b7...'
Uninstalling xone v0.3-88-gf6f30b7...
+ dkms remove -m xone -v v0.3-88-gf6f30b7 --all
Module xone v0.3-88-gf6f30b7 is not installed for kernel 6.11.0-rc4-7-MANJARO (x86_64). Skipping...
Deleting module xone-v0.3-88-gf6f30b7 completely from the DKMS tree.
+ rm -r /usr/src/xone-v0.3-88-gf6f30b7
+ rm -f /etc/modprobe.d/xone-blacklist.conf
+ rm -f /usr/local/bin/xone-get-firmware.sh
```

dlundqvist, 2024-09-08:
Now after `uninstall.sh` without installing again, what does `sudo dkms status` say?

mben42, 2024-09-08:
```[16:06:45] [~/Téléchargements/xone] [master]
-> % sudo dkms status
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/xone-v0.3-88/gf6f30b7/source/dkms.conf does not exist.
xone-v0.3-57/g29ec357: added```

Should I manually remove anything related to xone (I see much things when I run a locate)?

mben42, 2024-09-08:
```[16:00:11] [~/Téléchargements/xone] [master]
-> % locate xone                                                                    
/home/my_surname/Téléchargements/xone
/home/my_surname/Téléchargements/xone/.git
/home/my_surname/Téléchargements/xone/.github
/home/my_surname/Téléchargements/xone/.gitignore
/home/my_surname/Téléchargements/xone/Kbuild
/home/my_surname/Téléchargements/xone/LICENSE
/home/my_surname/Téléchargements/xone/README.md
/home/my_surname/Téléchargements/xone/auth
/home/my_surname/Téléchargements/xone/bus
/home/my_surname/Téléchargements/xone/dkms.conf
/home/my_surname/Téléchargements/xone/driver
/home/my_surname/Téléchargements/xone/install
/home/my_surname/Téléchargements/xone/install.sh
/home/my_surname/Téléchargements/xone/logo.svg
/home/my_surname/Téléchargements/xone/transport
/home/my_surname/Téléchargements/xone/uninstall.sh
/home/my_surname/Téléchargements/xone/.github/FUNDING.yml
/home/my_surname/Téléchargements/xone/auth/auth.c
/home/my_surname/Téléchargements/xone/auth/auth.h
/home/my_surname/Téléchargements/xone/auth/crypto.c
/home/my_surname/Téléchargements/xone/auth/crypto.h
/home/my_surname/Téléchargements/xone/bus/bus.c
/home/my_surname/Téléchargements/xone/bus/bus.h
/home/my_surname/Téléchargements/xone/bus/protocol.c
/home/my_surname/Téléchargements/xone/bus/protocol.h
/home/my_surname/Téléchargements/xone/driver/chatpad.c
/home/my_surname/Téléchargements/xone/driver/common.c
/home/my_surname/Téléchargements/xone/driver/common.h
/home/my_surname/Téléchargements/xone/driver/gamepad.c
/home/my_surname/Téléchargements/xone/driver/headset.c
/home/my_surname/Téléchargements/xone/driver/madcatz_glam.c
/home/my_surname/Téléchargements/xone/driver/madcatz_strat.c
/home/my_surname/Téléchargements/xone/driver/pdp_jaguar.c
/home/my_surname/Téléchargements/xone/install/firmware.sh
/home/my_surname/Téléchargements/xone/install/modprobe.conf
/home/my_surname/Téléchargements/xone/transport/dongle.c
/home/my_surname/Téléchargements/xone/transport/mt76.c
/home/my_surname/Téléchargements/xone/transport/mt76.h
/home/my_surname/Téléchargements/xone/transport/mt76_defs.h
/home/my_surname/Téléchargements/xone/transport/wired.c
/usr/lib/modules/6.11.0-rc4-7-MANJARO/updates/dkms/xone-dongle.ko.zst
/usr/lib/modules/6.11.0-rc4-7-MANJARO/updates/dkms/xone-gip-chatpad.ko.zst
/usr/lib/modules/6.11.0-rc4-7-MANJARO/updates/dkms/xone-gip-gamepad.ko.zst
/usr/lib/modules/6.11.0-rc4-7-MANJARO/updates/dkms/xone-gip-headset.ko.zst
/usr/lib/modules/6.11.0-rc4-7-MANJARO/updates/dkms/xone-gip-madcatz-glam.ko.zst
/usr/lib/modules/6.11.0-rc4-7-MANJARO/updates/dkms/xone-gip-madcatz-strat.ko.zst
/usr/lib/modules/6.11.0-rc4-7-MANJARO/updates/dkms/xone-gip-pdp-jaguar.ko.zst
/usr/lib/modules/6.11.0-rc4-7-MANJARO/updates/dkms/xone-gip.ko.zst
/usr/lib/modules/6.11.0-rc4-7-MANJARO/updates/dkms/xone-wired.ko.zst
/usr/src/xone-v0.3-57-g29ec357
/usr/src/xone-v0.3-88-gf6f30b7
/usr/src/xone-v0.3-57-g29ec357/.git
/usr/src/xone-v0.3-57-g29ec357/.github
/usr/src/xone-v0.3-57-g29ec357/.gitignore
/usr/src/xone-v0.3-57-g29ec357/Kbuild
/usr/src/xone-v0.3-57-g29ec357/LICENSE
/usr/src/xone-v0.3-57-g29ec357/README.md
/usr/src/xone-v0.3-57-g29ec357/auth
/usr/src/xone-v0.3-57-g29ec357/bus
/usr/src/xone-v0.3-57-g29ec357/dkms.conf
/usr/src/xone-v0.3-57-g29ec357/driver
/usr/src/xone-v0.3-57-g29ec357/install
/usr/src/xone-v0.3-57-g29ec357/install.sh
/usr/src/xone-v0.3-57-g29ec357/logo.svg
/usr/src/xone-v0.3-57-g29ec357/transport
/usr/src/xone-v0.3-57-g29ec357/uninstall.sh
/usr/src/xone-v0.3-57-g29ec357/.github/FUNDING.yml
/usr/src/xone-v0.3-57-g29ec357/auth/auth.c
[... 99 lines cut ...]
/var/lib/dkms/xone-v0.3-57/g29ec357/build/transport/mt76.c
/var/lib/dkms/xone-v0.3-57/g29ec357/build/transport/mt76.h
/var/lib/dkms/xone-v0.3-57/g29ec357/build/transport/mt76.o
/var/lib/dkms/xone-v0.3-57/g29ec357/build/transport/mt76_defs.h
/var/lib/dkms/xone-v0.3-57/g29ec357/build/transport/wired.c
/var/lib/dkms/xone-v0.3-57/g29ec357/build/transport/wired.o
/var/lib/dkms/xone-v0.3-88/gf6f30b7
/var/lib/dkms/xone-v0.3-88/kernel-6.11.0-rc4-7-MANJARO-x86_64
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO
/var/lib/dkms/xone-v0.3-88/gf6f30b7/source
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/log
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/module
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/log/.config
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/log/make.log
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/module/xone-dongle.ko.zst
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/module/xone-gip-chatpad.ko.zst
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/module/xone-gip-gamepad.ko.zst
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/module/xone-gip-headset.ko.zst
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/module/xone-gip-madcatz-glam.ko.zst
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/module/xone-gip-madcatz-strat.ko.zst
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/module/xone-gip-pdp-jaguar.ko.zst
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/module/xone-gip.ko.zst
/var/lib/dkms/xone-v0.3-88/gf6f30b7/6.11.0-rc4-7-MANJARO/x86_64/module/xone-wired.ko.zst
```

dlundqvist, 2024-09-08:
For starters I think just moving away `/var/lib/dkms/xone-v0.3-88` should progress a bit more.

dlundqvist, 2024-09-08:
After that run `sudo dkms status` again. It should give you status about all DKMS installed kernel modules.

mben42, 2024-09-08:
```zsh
[16:18:05] [~/Téléchargements/xone] [master]
-> % sudo dkms status                               
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/xone-v0.3-88/gf6f30b7/source/dkms.conf does not exist.
```

Then I tried to install again and got the same problem:
```zsh
[16:18:39] [~/Téléchargements/xone] [master]
-> % sudo sh -x ./install.sh 
+ set -eu
++ id -u
+ '[' 0 -ne 0 ']'
++ command -v dkms
+ '[' -x /usr/bin/dkms ']'
++ dkms status xone
+ '[' -n '' ']'
+ '[' -f /usr/local/bin/xow ']'
+ '[' -n mathis ']'
++ sudo -u mathis git describe --tags
+ version=v0.3-88-gf6f30b7
+ source=/usr/src/xone-v0.3-88-gf6f30b7
+ log=/var/lib/dkms/xone/v0.3-88-gf6f30b7/build/make.log
+ echo 'Installing xone v0.3-88-gf6f30b7...'
Installing xone v0.3-88-gf6f30b7...
+ cp -r . /usr/src/xone-v0.3-88-gf6f30b7
+ find /usr/src/xone-v0.3-88-gf6f30b7 -type f '(' -name dkms.conf -o -name '*.c' ')' -exec sed -i s/#VERSION#/v0.3-88-gf6f30b7/ '{}' +
+ '[' '' '!=' --release ']'
+ echo 'ccflags-y += -DDEBUG'
+ dkms install -m xone -v v0.3-88-gf6f30b7
Sign command: /usr/lib/modules/6.11.0-rc4-7-MANJARO/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/xone/v0.3-88-gf6f30b7/source -> /usr/src/xone-v0.3-88-gf6f30b7

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.11.0-rc4-7-MANJARO -C /usr/lib/modules/6.11.0-rc4-7-MANJARO/build M=/var/lib/dkms/xone/v0.3-88-gf6f30b7/build...
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-wired.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-dongle.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip-gamepad.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip-headset.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip-chatpad.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip-madcatz-strat.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip-madcatz-glam.ko
Signing module /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/xone-gip-pdp-jaguar.ko
Cleaning build area...

xone-wired.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-wired.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-dongle.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-dongle.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip-gamepad.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip-gamepad.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip-headset.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip-headset.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip-chatpad.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip-chatpad.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip-madcatz-strat.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip-madcatz-strat.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip-madcatz-glam.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip-madcatz-glam.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.

xone-gip-pdp-jaguar.ko.zst:
Running module version sanity check.
Module version v0.3-88-gf6f30b7 for xone-gip-pdp-jaguar.ko.zst
exactly matches what is already found in kernel 6.11.0-rc4-7-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.
Error! Installation aborted.
+ '[' -r /var/lib/dkms/xone/v0.3-88-gf6f30b7/build/make.log ']'
+ exit 1
```

dlundqvist, 2024-09-08:
Hmm... is your clone ok? Could you do `git status` and `git log --oneline -n 5`?

mben42, 2024-09-08:
git status tells branch is up to date and working directory is clean:
```zsh
[16:22:19] [~/Téléchargements/xone] [master]
-> % git status
Sur la branche master
Votre branche est à jour avec 'origin/master'.

rien à valider, la copie de travail est propre
```

`git log --oneline -n 5` output:
```zsh
f6f30b7 (HEAD -> master, origin/master, origin/HEAD) Merge remote-tracking branch 'garethp/master'
c994fea Merge remote-tracking branch 'tskaar/fix-6.11'
e3cef7b Add a Vendor ID for Logitech Astro headphones
28df566 fix: build on kernel v6.11
c7b24b7 bus: Remove unnecessary compatibility remove callback for 5.15.
```

dlundqvist, 2024-09-08:
I'm sorry it's not clear to me what the problem is. Without installing Manjaro myself I don't have anything besides removing everything `xone` related outside of `$HOME` to suggest.

mben42, 2024-09-08:
> I'm sorry it's not clear to me what the problem is. Without installing Manjaro myself I don't have anything besides removing everything `xone` related outside of `$HOME` to suggest.

Thanks for your help anyway, it did work, I was able to install and my controller works!

closed 2024-09-08 by mben42

dlundqvist, 2024-09-08:
Great! Although it annoys me that we don't know what the actual problem was, good that you finally got it working so you can stop hanging around here and get some fun done.