dlundqvist/xone #14: Hard freeze/hang with one of the gip modules (unknown which one yet)

GloriousEggroll, 2024-09-24:
Reporting this here as it seems a bit more active at least until upstream work starts again.

One of the gip modules is causing a hard hang/freeze when rapidly unplugging + replugging via usb cable. Tested on xbox elite v2 controller.  I tested this 3 different ways:
1. xone driver only
2. xone driver + xpad-noone driver
3. xone driver with xone_wired module removed (dongle support only) + xpad module re-enabled

After slimming down the loaded modules to just these:

xone_dongle
xone_gip_gamepad
xone_gip

unplugging/plugging rapidly works fine. I'll dig more later into this to try to see which module is causing the issue.

dlundqvist, 2024-09-24:
Thanks for investigating. I'll see what I can do once we know more.

Lawstorant, 2025-07-01:
@GloriousEggroll is this something that's still happening? I was unable to reproduce with my controllers though I don't have the Elite.

Lawstorant, 2025-08-02:
Got myself a Series 2 Elite controller. Can't reproduce

GloriousEggroll, 2026-03-20:
If this makes sense to you I can open an MR, just lmk