dlundqvist/xone #15: 6.11 kernel: Thrustmaster eSwap not working

lostiniceland, 2024-10-11:
Hi switched to your fork due to the 6.11 kernel issues.

The fixes solved the build-errors but the controller ist not working via usb cable.

`lsusb` shows the usb device, but `evtest` for instance does not show the device.

I've installed the module with debug symbols and used `sudo dmesg` to see whats going on.
There is one interesting line `Loading of module with unavailable key is rejected` which indicates the module is not loaded.
```
[ 8802.400597] [  T16277] usb 7-2: USB disconnect, device number 3
[ 8804.444300] [  T27032] usb 7-2: new full-speed USB device number 4 using xhci_hcd
[ 8804.613073] [  T27032] usb 7-2: New USB device found, idVendor=044f, idProduct=d012, bcdDevice= 1.00
[ 8804.613077] [  T27032] usb 7-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8804.613078] [  T27032] usb 7-2: Product: eSwap PRO Controller Xbox
[ 8804.613080] [  T27032] usb 7-2: Manufacturer: ThrustMaster
[ 8804.613081] [  T27032] usb 7-2: SerialNumber: 0000F199FCE53707
[ 8804.631010] [  T27028] Loading of module with unavailable key is rejected

```

dlundqvist, 2024-10-13:
That looks like a problem with Secure Boot. There's a pinned message on xone's Discord server with more information: https://discord.com/channels/733964971842732042/819644054563192846/1119974224833024040.

lostiniceland, 2024-10-14:
Not sure if this is really necessary.
There were some problems with this controller at the end of February this year where I've supplied a Wireshark capture.
This was the discussion: https://discord.com/channels/733964971842732042/1203016473317085184

Those commits seem to be the ones that fixed the issue.
https://github.com/medusalix/xone/commit/87f538ca0c31f81ed333e7e898b6ed4d1cc03d9c
https://github.com/medusalix/xone/commit/2dd7129416757c2a08548c939404ab55df8a0b67
https://github.com/medusalix/xone/commit/9872e24d9c8eb796d71cc5e821cac741d1b1ea83

dlundqvist, 2024-10-15:
Those three commits are in my fork. I'm not sure what you mean but `Loading of module with unavailable key is rejected` does indicate this is a problem with SecureBoot. Do you perhaps have other logs to share?

lostiniceland, 2024-10-16:
I am going to try to figure out whats really going on here. The dmesg entry with the failed key has a different ID (the T-thingy) than the device. But the message is always showing up when plugin the device in.
Maybe there is something with Opensuse and how they handly modules.

lostiniceland, 2024-10-16:
You were right about the signing. I was a bit hesitant because this was not necessary before.
After enrolling the mok the module was loaded.

closed 2024-10-16 by lostiniceland