dlundqvist/xone #17: Build failure on Kernel 6.12

devz3r0, 2024-11-19:
Hi,

I am using manjaro with kernel 6.12 and got the following error:

```
Installing xone v0.3-90-gf314bce...

Sign command: /usr/lib/modules/6.12.0-1-MANJARO/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/xone/v0.3-90-gf314bce/source -> /usr/src/xone-v0.3-90-gf314bce

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j8 KERNELRELEASE=6.12.0-1-MANJARO -C /usr/lib/modules/6.12.0-1-MANJARO/build M=/var/lib/dkms/xone/v0.3-90-gf314bce/build
Error! Bad return status for module build on kernel: 6.12.0-1-MANJARO (x86_64)
Consult /var/lib/dkms/xone/v0.3-90-gf314bce/build/make.log for more information.
DKMS make.log for xone-v0.3-90-gf314bce for kernel 6.12.0-1-MANJARO (x86_64)
Di 19 Nov 2024 17:13:11 CET
make: Entering directory '/usr/lib/modules/6.12.0-1-MANJARO/build'
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/transport/wired.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/transport/dongle.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/transport/mt76.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/bus/bus.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/bus/protocol.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/auth/auth.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/auth/crypto.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/common.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/gamepad.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/headset.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/chatpad.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/madcatz_strat.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/madcatz_glam.o
  CC [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/pdp_jaguar.o
  LD [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/xone-wired.o
  LD [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/xone-gip.o
/var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/headset.c: In function ‘gip_headset_pcm_hw_params’:
/var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/headset.c:93:16: error: implicit declaration of function ‘snd_pcm_lib_alloc_vmalloc_buffer’ [-Wimplicit-function-declaration]
   93 |         return snd_pcm_lib_alloc_vmalloc_buffer(sub,
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  LD [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/xone-gip-gamepad.o
  LD [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/xone-gip-madcatz-strat.o
  LD [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/xone-gip-madcatz-glam.o
/var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/headset.c: In function ‘gip_headset_pcm_hw_free’:
/var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/headset.c:99:16: error: implicit declaration of function ‘snd_pcm_lib_free_vmalloc_buffer’; did you mean ‘snd_pcm_lib_malloc_pages’? [-Wimplicit-function-declaration]
   99 |         return snd_pcm_lib_free_vmalloc_buffer(sub);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                snd_pcm_lib_malloc_pages
  LD [M]  /var/lib/dkms/xone/v0.3-90-gf314bce/build/xone-dongle.o
/var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/headset.c: At top level:
/var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/headset.c:160:17: error: ‘snd_pcm_lib_get_vmalloc_page’ undeclared here (not in a function); did you mean ‘snd_pcm_lib_malloc_pages’?
  160 |         .page = snd_pcm_lib_get_vmalloc_page,
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                 snd_pcm_lib_malloc_pages
make[2]: *** [scripts/Makefile.build:229: /var/lib/dkms/xone/v0.3-90-gf314bce/build/driver/headset.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/lib/modules/6.12.0-1-MANJARO/build/Makefile:1936: /var/lib/dkms/xone/v0.3-90-gf314bce/build] Error 2
make: *** [Makefile:224: __sub-make] Error 2
make: Leaving directory '/usr/lib/modules/6.12.0-1-MANJARO/build'
```

pmatos, 2024-11-20:
If you don't care about headset, try this patch:

```
diff --git a/Kbuild b/Kbuild
index 37cdc0e..fb65943 100644
--- a/Kbuild
+++ b/Kbuild
@@ -2,9 +2,8 @@ xone-wired-y := transport/wired.o
 xone-dongle-y := transport/dongle.o transport/mt76.o
 xone-gip-y := bus/bus.o bus/protocol.o auth/auth.o auth/crypto.o driver/common.o
 xone-gip-gamepad-y := driver/gamepad.o
-xone-gip-headset-y := driver/headset.o
 xone-gip-chatpad-y := driver/chatpad.o
 xone-gip-madcatz-strat-y := driver/madcatz_strat.o
 xone-gip-madcatz-glam-y := driver/madcatz_glam.o
 xone-gip-pdp-jaguar-y := driver/pdp_jaguar.o
-obj-m := xone-wired.o xone-dongle.o xone-gip.o xone-gip-gamepad.o xone-gip-headset.o xone-gip-chatpad.o xone-gip-madcatz-strat.o xone-gip-madcatz-glam.o xone-gip-pdp-jaguar.o
+obj-m := xone-wired.o xone-dongle.o xone-gip.o xone-gip-gamepad.o xone-gip-chatpad.o xone-gip-madcatz-strat.o xone-gip-madcatz-glam.o xone-gip-pdp-jaguar.o
diff --git a/dkms.conf b/dkms.conf
index b6131c2..0a7b01e 100644
--- a/dkms.conf
+++ b/dkms.conf
@@ -4,11 +4,10 @@ BUILT_MODULE_NAME[0]="xone-wired"
 BUILT_MODULE_NAME[1]="xone-dongle"
 BUILT_MODULE_NAME[2]="xone-gip"
 BUILT_MODULE_NAME[3]="xone-gip-gamepad"
-BUILT_MODULE_NAME[4]="xone-gip-headset"
-BUILT_MODULE_NAME[5]="xone-gip-chatpad"
-BUILT_MODULE_NAME[6]="xone-gip-madcatz-strat"
-BUILT_MODULE_NAME[7]="xone-gip-madcatz-glam"
-BUILT_MODULE_NAME[8]="xone-gip-pdp-jaguar"
+BUILT_MODULE_NAME[4]="xone-gip-chatpad"
+BUILT_MODULE_NAME[5]="xone-gip-madcatz-strat"
+BUILT_MODULE_NAME[6]="xone-gip-madcatz-glam"
+BUILT_MODULE_NAME[7]="xone-gip-pdp-jaguar"
 DEST_MODULE_LOCATION[0]="/kernel/drivers/input/joystick"
 DEST_MODULE_LOCATION[1]="/kernel/drivers/input/joystick"
 DEST_MODULE_LOCATION[2]="/kernel/drivers/input/joystick"
@@ -17,5 +16,4 @@ DEST_MODULE_LOCATION[4]="/kernel/drivers/input/joystick"
 DEST_MODULE_LOCATION[5]="/kernel/drivers/input/joystick"
 DEST_MODULE_LOCATION[6]="/kernel/drivers/input/joystick"
 DEST_MODULE_LOCATION[7]="/kernel/drivers/input/joystick"
-DEST_MODULE_LOCATION[8]="/kernel/drivers/input/joystick"
 AUTOINSTALL="yes"
```

devz3r0, 2024-11-22:
Indeed, I do not have the headset, after applying your patch, the controller is working again. Thanks very much, however I assume that the thread should be kept open because the main issue is still there, correct?

pmatos, 2024-11-22:
> Indeed, I do not have the headset, after applying your patch, the controller is working again. Thanks very much, however I assume that the thread should be kept open because the main issue is still there, correct?

Right.

markmb88, 2024-11-30:
> If you don't care about headset, try this patch:
>
> ```
> diff --git a/Kbuild b/Kbuild
> index 37cdc0e..fb65943 100644
> --- a/Kbuild
> +++ b/Kbuild
> @@ -2,9 +2,8 @@ xone-wired-y := transport/wired.o [...]

bandaid works. Thanks!

Owewolf, 2024-12-22:
I get exactly the same on Ultramarine 6.12 which file do I patch

devz3r0, 2024-12-22:
I suggest you copy the xone directory, in that one you create a file like: dkms.conf.patch

write the following in:

```
index b6131c2..0a7b01e 100644
--- a/dkms.conf
+++ b/dkms.conf
@@ -4,11 +4,10 @@ BUILT_MODULE_NAME[0]="xone-wired"
 BUILT_MODULE_NAME[1]="xone-dongle"
 BUILT_MODULE_NAME[2]="xone-gip"
 BUILT_MODULE_NAME[3]="xone-gip-gamepad"
-BUILT_MODULE_NAME[4]="xone-gip-headset"
-BUILT_MODULE_NAME[5]="xone-gip-chatpad"
-BUILT_MODULE_NAME[6]="xone-gip-madcatz-strat"
-BUILT_MODULE_NAME[7]="xone-gip-madcatz-glam"
-BUILT_MODULE_NAME[8]="xone-gip-pdp-jaguar"
+BUILT_MODULE_NAME[4]="xone-gip-chatpad"
+BUILT_MODULE_NAME[5]="xone-gip-madcatz-strat"
+BUILT_MODULE_NAME[6]="xone-gip-madcatz-glam"
+BUILT_MODULE_NAME[7]="xone-gip-pdp-jaguar"
 DEST_MODULE_LOCATION[0]="/kernel/drivers/input/joystick"
 DEST_MODULE_LOCATION[1]="/kernel/drivers/input/joystick"
 DEST_MODULE_LOCATION[2]="/kernel/drivers/input/joystick"
@@ -17,5 +16,4 @@ DEST_MODULE_LOCATION[4]="/kernel/drivers/input/joystick"
 DEST_MODULE_LOCATION[5]="/kernel/drivers/input/joystick"
 DEST_MODULE_LOCATION[6]="/kernel/drivers/input/joystick"
 DEST_MODULE_LOCATION[7]="/kernel/drivers/input/joystick"
-DEST_MODULE_LOCATION[8]="/kernel/drivers/input/joystick"
 AUTOINSTALL="yes"
```

then execute:

```
patch dkms.conf < dkms.conf.patch
```

Tweekism, 2024-12-22:
If you pull the lastest version from here which is `v0.3-92-gaa6374a` it should already have a fix for kernel 6.12.

devz3r0, 2024-12-22:
so it is fixed?

Tweekism, 2024-12-22:
Yes, compared to your original post, which is now 2 commits behind current master. If you were to do a pull now you'll get the 6.12 fix from tskaar that fixed it for me.

devz3r0, 2024-12-23:
Thanks, then I'll close the issue.

closed 2024-12-23 by devz3r0