dlundqvist/xone #21: Build succeeds but Dongle not working (Kernel 6.11.10)

monopauli, 2024-12-13:
Building on Fedora 41 (Kernel 6.11.10). Builds successfully but the wireless dongle does not seem to activate. Any more steps I should take? 

```
Installing xone v0.3-92-gaa6374a...

Sign command: /lib/modules/6.11.10-300.fc41.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/xone/v0.3-92-gaa6374a/source -> /usr/src/xone-v0.3-92-gaa6374a

Cleaning build area... done.
Building module(s)... done.
Signing module /var/lib/dkms/xone/v0.3-92-gaa6374a/build/xone-wired.ko
Signing module /var/lib/dkms/xone/v0.3-92-gaa6374a/build/xone-dongle.ko
Signing module /var/lib/dkms/xone/v0.3-92-gaa6374a/build/xone-gip.ko
Signing module /var/lib/dkms/xone/v0.3-92-gaa6374a/build/xone-gip-gamepad.ko
Signing module /var/lib/dkms/xone/v0.3-92-gaa6374a/build/xone-gip-headset.ko
Signing module /var/lib/dkms/xone/v0.3-92-gaa6374a/build/xone-gip-chatpad.ko
Signing module /var/lib/dkms/xone/v0.3-92-gaa6374a/build/xone-gip-madcatz-strat.ko
Signing module /var/lib/dkms/xone/v0.3-92-gaa6374a/build/xone-gip-madcatz-glam.ko
Signing module /var/lib/dkms/xone/v0.3-92-gaa6374a/build/xone-gip-pdp-jaguar.ko
Cleaning build area... done.

Installing /lib/modules/6.11.10-300.fc41.x86_64/extra/xone-wired.ko.xz
Installing /lib/modules/6.11.10-300.fc41.x86_64/extra/xone-dongle.ko.xz
Installing /lib/modules/6.11.10-300.fc41.x86_64/extra/xone-gip.ko.xz
Installing /lib/modules/6.11.10-300.fc41.x86_64/extra/xone-gip-gamepad.ko.xz
Installing /lib/modules/6.11.10-300.fc41.x86_64/extra/xone-gip-headset.ko.xz
Installing /lib/modules/6.11.10-300.fc41.x86_64/extra/xone-gip-chatpad.ko.xz
Installing /lib/modules/6.11.10-300.fc41.x86_64/extra/xone-gip-madcatz-strat.ko.xz
Installing /lib/modules/6.11.10-300.fc41.x86_64/extra/xone-gip-madcatz-glam.ko.xz
Installing /lib/modules/6.11.10-300.fc41.x86_64/extra/xone-gip-pdp-jaguar.ko.xz
Running depmod.... done.
```

Firmware also seems to be present: 
```
The firmware for the wireless dongle is subject to Microsoft's Terms of Use:
https://www.microsoft.com/en-us/legal/terms-of-use

Press enter to continue!

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  195k  100  195k    0     0   221k      0 --:--:-- --:--:-- --:--:--  221k
driver.cab: WARNING; possible 16256 extra bytes at end of file.
Extracting cabinet: driver.cab
  extracting FW_ACC_00U.bin

All done, no errors.
FW_ACC_00U.bin: OK
```

lynxbites, 2024-12-16:
Same issue here but on 6.12.4

slice22358, 2024-12-23:
Same problem, Fedora 41, 6.12.5

Tweekism, 2024-12-25:
Have you tried asking in the xone channel on Discord? NoXPhasma there is really helpful with the troubleshooting.

https://discord.gg/cnjwbWfaBN

VonSparq, 2025-01-20:
Same here on Kubuntu 24.10, kernel 6.11.0-13-generic. I tried this version, as well as [medusalix](https://github.com/medusalix)'s one which at first didn't build but I tried these two PSs but neither of them worked.
https://github.com/medusalix/xone/pull/56
https://github.com/medusalix/xone/pull/48

The issue is as you described, the package is built and installed properly and the firmware is also downloaded without errors, but the dongle just doesn't react to the button and the pad doesn't connect. Rebooting doesn't help, at least the pad works fine with the cable.

Lawstorant, 2025-04-16:
@VonSparq this might be related: https://github.com/dlundqvist/xone/pull/49

tthe3d, 2025-06-17:
I ChatGPT-ed the make logs.
Uninstalled xone.
did : sudo apt install shim-signed
Installed again 
Reboot and it worked