dlundqvist/xone #23: Can't install because "driver is already installed" while it's not.

jmfergeau, 2024-12-17:
Trying to install it manually for the 6.12 patch but it simply doesn't work. It says that `Driver is already installed!`. Though, it's completely uninstalled everywhere.

When I try to uninstall, i'm just getting this:

```
Unloading modules: xone_wired xone_gip ...
modprobe: WARNING: Module xone_wired not found.
modprobe: WARNING: Module xone_gip not found.
```

Thanks

Tweekism, 2024-12-22:
Hey mate, did you get this sorted?

Are you currently running a 6.12 kernel or are you trying to upgrade to 6.12.x? If the latter, then the easiest fix would probably be to go to your /usr/src directory and delete any 'xone' directories there. Then upgrade your kernel, reboot and then reinstall xone as normal.

jmfergeau, 2024-12-22:
I sorted it out by manually installing this version that also fixes support for kernel 6.12

https://github.com/tskaar/xone/tree/fix-6.12

closed 2024-12-22 by jmfergeau