dlundqvist/xone #24: Issues with using multiple Xbox Series controllers (names get mixed up, connection drops)

OverdoneAndDry, 2025-01-03:
Not sure if these are Xone issue or Steam Deck issues, but I'm trying to connect 4 Xbox Series controllers to my Steam Deck via Xone and the official Xbox Wireless Adapter. I can get the driver installed and most of the time, I can connect to the Xbox Series controllers okay. Sometimes they don't connect but if I unplug and re-plug the dongle and/or restart the Deck, I can get them to connect.

However, I run into 2 issues that I can't find a fix for.

1st is pretty minor but not sure if it's related to the second issue. The Deck has trouble keeping track of the Xbox Series controller names when they have the same firmware (updated via the Xbox Accessories app on Windows). When I originally tried connecting my Xbox controllers to the Deck, they all had different firmware and the Deck was able to differentiate between them (I had names like "Xbox White" and "Xbox Pink"). But when I updated the firmware so they're all up-to-date, the names get all mixed up. Both my white and pink controller show up as "Xbox White" for example.

2nd issue is a bigger problem. I get disconnection issues when I have multiple controllers connected. I'm trying to see if 1 controller is rock solid or if it might still disconnect. But when I have 4 controllers connected to the dongle, sometimes inputs lock up and a controller gets stuck permanently pressing "right" or something. I didn't have a chance to do much testing in a local multiplayer setting, but I can try to test more specific things in the coming weeks if it might help. From the couple instances I've tried, usually if there's 1 disconnect or hiccup, the connection remains unstable until I just restart the Deck.

To compare, I have 4 Xbox 360 controllers that I can connect to the Deck via the old Xbox 360 wireless dongle and those connections are rock solid.

dlundqvist, 2025-03-12:
Is it only Steam Deck that has problem? Or also Xbox Accessories in Windows? Do you have a way of testing the dongle on something Linux based besides Steam Deck?

Lawstorant, 2025-08-03:
I now have Xbox One controller, 2 Series controllers and Elite Series 2 so I could look into this

OverdoneAndDry, 2025-08-09:
Thanks for the replies dlundqvist and Lawstorant. So, after having problems with the dongle, I decided to buy another one. I bought the 1st one from an Amazon 3rd party seller and looks like the official OEM one. I got the 2nd from eBay, with the seller also claiming it's official. It looks identical to the 1st one, but everything is rock solid on the Steam Deck with it. I haven't tried mixing up multiple controllers though. I've been a bit lazy about that and haven't had the chance to experiment. If you're interested, I can try that out and report back.

Lawstorant, 2025-11-07:
There isn't really anything we can do at the drier level to make steam recognize the controllers :/ I'll try to bug steam guys in the future as they're looking at the BT controller MAC manually so anything I add, won't matter for now.