dlundqvist/xone #29: Xbox wireless dongle and Elite controller too

jsphonorio, 2025-01-22:
I am running beside which uses zone and noticed that while I can connect my Elite to the wireless dongle and have all the buttons working and having it come up with the proper name of an elite, the back paddles are not being recognized. I'm on the latest firmware, and if I use one of my profiles saved in Windows the back pedal will operate since they're just being mapped directly to specific buttons already but in the default blank profile they don't even appear in the controller tester or Steam.

Entropy512, 2025-02-13:
There's an open PR to the original project to add back paddle support for some firmware versions not yet supported, although the original submitter of that PR seems to have disappeared.

I'm going to try to get around this weekend to rebasing my additions to that PR and submit it here.

https://github.com/Entropy512/xone/tree/paddles is functional but needs cleanup.

https://github.com/medusalix/xone/pull/41 is the original PR with some discussion

Doty1154, 2025-04-07:
love to see this project is getting more regular work. itd be awesome if this patch for the paddles got merged, but i wonder what has changed since that devs work, if newer firmware is still compatible etc.

charlesmunger, 2025-04-29:
Sorry for disappearing, I don't have windows to test, so I couldn't really do anything with newer firmware - and it's not clear what benefit the newer firmware offers, so I have just been running with my own patch built locally.

Hassiopeia, 2025-05-27:
Is there any progress or update on moving the PR into production?
https://github.com/dlundqvist/xone/issues/29#issuecomment-2657591787

This is really the last thing keeping me from switching to Linux on my gaming PC.