dlundqvist/xone #30: Wooting 60he support

JustCauseWhyNot, 2025-01-28:
I want to know if the [xinput mode](https://help.wooting.io/article/93-configuring-xinput-support-for-linux) offered by the wooting 60he providing analog input on a keyboard could be supported by xone? Currently xone doesn't interact with it if I try to use it as a gamepad. Xpad does work with it. But I'm trying to figure out if I wooting needs to add gip support. Or if it will already work with xone, and just requiring wooting to be added as a supported vendor.

dlundqvist, 2025-03-12:
Not sure exactly how that would work but xone does support what it calls "chatpad." I don't know if that is enough here. But as start, adding wooting as a vendor won't hurt. Do you know how to do that by yourself to test?

JustCauseWhyNot, 2025-03-12:
>Do you know how to do that by yourself to test?

I think so. I've created a patch file adding wootings vendor id to transport/wired.c, and I'll try building it to test it out.

JustCauseWhyNot, 2025-03-13:
Here's the patch I made. <summary> wootingvendor.patch </summary>
<details>

```
diff --git a/transport/wired.c b/transport/wired.c
index 1872f31..79a9851 100644
--- a/transport/wired.c
+++ b/transport/wired.c
@@ -567,6 +567,7 @@ static const struct usb_device_id xone_wired_id_table[] = {
        { XONE_WIRED_VENDOR(0x046d) }, /* Logitech Astro */
        { XONE_WIRED_VENDOR(0x0079) }, /* EasySMX */
        { XONE_WIRED_VENDOR(0x1038) }, /* SteelSeries ApS */
+       { XONE_WIRED_VENDOR(0x31e3) }, /* Wooting */
        { },
 };
```

</details>

Unless I missed another thing I needed to do. Adding just wooting to the vendor list unfortunately wasn't enough to get the keyboards gamepad inputs to work with the xone driver.

Lawstorant, 2025-11-16:
Wooting exposes Xbox 360 controller, not a GIP (Xbox one+) controller. Xone doesn't support pre gip devices