dlundqvist/xone #31: 6.13 no input detected

lostiniceland, 2025-01-31:
With Linux 6.13 the build succeeds but no input is recognized.

Tested with openSuse Tumbleweed and Thrustmaster controller.
The controller shows up in KDEs system settings but no axis nor buttons receive any input.

PadTrick, 2025-02-04:
archlinux 6.13.zen1-2 works for me without issues (xbox series x controller) with dongle & cabel.
but on 6.13.1.zen1-1 dkms install fails

lostiniceland, 2025-02-04:
Then there must be some difference with the Thrustmaster eSwap model.
I've tried your reconnect approach without success.

Slushnas, 2025-03-11:
Are there any xone related errors in your `sudo dmesg` output after connecting the controller?

dlundqvist, 2025-03-12:
Since this is 6.13 perhaps the problem was #33. Please try HEAD and see if problem is solved.

lostiniceland, 2025-03-12:
> Are there any xone related errors in your `sudo dmesg` output after connecting the controller?

`[  173.577928] [    T174] xone-gip-gamepad gip0.0: gip_auth_exchange_rsa: encrypt RSA failed: -17`

> Since this is 6.13 perhaps the problem was [#33](https://github.com/dlundqvist/xone/issues/33). Please try HEAD and see if problem is solved.

Yes, the error message stated above was the same as #33 and pulling your HEAD solved the issue. Many thanks.

closed 2025-03-12 by lostiniceland