dlundqvist/xone #33: Update to Linux 6.13.2-arch1-1 broke

BloomyInDev, 2025-02-12:
Hello,
I recently updated to kernel `6.13.2` and found out that my controller wasn't working anymore.

[Here is the logs in `dmesg`](https://github.com/user-attachments/files/18772648/controllerproblem.txt)

Yes i know this isn't an official controller but it is certified by Microsoft so i think it should work

I've tested on another machine with kernel `6.12.10` and it still works

If you have any questions, i'm free to help

dalailahner, 2025-02-13:
same here with my Turtle Beach Stealth Ultra.
dmesg:
```
[  762.659707] usb 1-3: new full-speed USB device number 7 using xhci_hcd
[  763.024440] usb 1-3: New USB device found, idVendor=10f5, idProduct=7070, bcdDevice= 1.05
[  763.024444] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  763.024446] usb 1-3: Product: Turtle Beach Stealth Ultra
[  763.024447] usb 1-3: Manufacturer: Turtle Beach
[  763.024449] usb 1-3: SerialNumber: 00000101B83DCE6B
[  763.214787] usb 1-3: reset full-speed USB device number 7 using xhci_hcd
[  764.009714] input: Microsoft Xbox Controller as /devices/pci0000:00/0000:00:02.1/0000:02:00.0/0000:03:0c.0/0000:0b:00.0/usb1/1-3/1-3:1.0/gip0/gip0.0/input/input29
[  764.391796] xone-gip-gamepad gip0.0: gip_auth_exchange_rsa: encrypt RSA failed: -17
[  764.945580] input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input3
```

captainjamesflint, 2025-02-14:
Hi, I'm also coming across this issue with my PDP Jaguar Rock Band 4 Guitar controller.
The messages I'm seeing are similar to the above:
```
Feb 14 14:48:07 tar-valon kernel: xone-dongle 5-10:1.0: xone_dongle_toggle_pairing: enabled=1
Feb 14 14:48:12 tar-valon kernel: xone-dongle 5-10:1.0: xone_dongle_remove_client: wcid=1, address=7e:ed:8a:e3:fd:12
Feb 14 14:48:12 tar-valon kernel: xone-gip gip0: gip_destroy_adapter: unregistered
Feb 14 14:48:12 tar-valon kernel: xone-dongle 5-10:1.0: xone_dongle_pair_client: address=7e:ed:8a:e3:fd:12
Feb 14 14:48:12 tar-valon kernel: xone-dongle 5-10:1.0: xone_dongle_toggle_pairing: enabled=0
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_create_adapter: registered
Feb 14 14:48:13 tar-valon kernel: xone-dongle 5-10:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:8a:e3:fd:12
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_get_client: initialized client 0
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:8a:e3:fd:12, vendor=0x0e6f, product=0x0170
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_handle_pkt_announce: firmware=1.0.10.1, hardware=257.1.1.1
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x00db
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x003a
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x00ae, length=0x002d
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x00db, length=0x0000
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_parse_client_commands: command=0x20, length=0x0e, options=0x14
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_parse_client_commands: command=0x21, length=0x05, options=0x0c
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_parse_firmware_versions: version=1.0
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_parse_capabilities: out=01-02-03-04-06-07
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_parse_capabilities: in=01-04-05-06-0a
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_parse_classes: class=PDP.Xbox.Guitar.Jaguar
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_parse_classes: class=Windows.Xbox.Input.NavigationController
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_parse_interfaces: guid=f66a261a-463a-e345-b9b6-0f2c0b2c1ebe
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_parse_interfaces: guid=e71ff3b8-8673-e940-a9f8-2f21263acfb7
Feb 14 14:48:13 tar-valon kernel: xone-gip gip0: gip_parse_interfaces: guid=56ff7697-fd9b-8145-ad45-b645bba526d6
Feb 14 14:48:13 tar-valon kernel: xone-gip-pdp-jaguar gip0.0: gip_register_client: registered
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_init_chunk_buffer: command=0x06, length=0x005a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x0020
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x005a, length=0x0000
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_init_chunk_buffer: command=0x06, length=0x0339
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x00ae, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x00e8, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x0122, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x015c, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x0196, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x01d0, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x020a, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x0244, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x027e, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x02b8, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x02f2, length=0x003a
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x032c, length=0x000d
Feb 14 14:48:14 tar-valon kernel: xone-gip gip0: gip_process_pkt_chunked: offset=0x0339, length=0x0000
Feb 14 14:48:14 tar-valon kernel: xone-gip-pdp-jaguar gip0.0: gip_auth_exchange_rsa: encrypt RSA failed: -17
```

justinlicitis, 2025-02-14:
I'm sure it's all the same issue, and sorry if this is redundant, but I'm getting the same with an 8BitDo Ultimate for Xbox. My brand name Xbox X|S works perfectly on the same machine. 
```
[ 1898.517356] usb 1-4: new full-speed USB device number 9 using xhci_hcd
[ 1898.643412] usb 1-4: New USB device found, idVendor=2dc8, idProduct=2016, bcdDevice= 1.14
[ 1898.643428] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1898.643436] usb 1-4: Product: 8BitDo Ultimate Wired Controller for Xbox
[ 1898.643442] usb 1-4: Manufacturer: 8BitDo
[ 1898.643448] usb 1-4: SerialNumber: 00002B588AA74D08
[ 1898.757687] usb 1-4: reset full-speed USB device number 9 using xhci_hcd
[ 1898.883104] xone-gip gip0: gip_create_adapter: registered
[ 1898.899590] xone-gip gip0: gip_get_client: initialized client 0
[ 1898.899594] xone-gip gip0: gip_handle_pkt_announce: address=08:4d:a7:8a:58:2b, vendor=0x2dc8, product=0x2016
[ 1898.899597] xone-gip gip0: gip_handle_pkt_announce: firmware=1.7.1584.84, hardware=1803.1.1.1
[ 1898.907639] xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x00f4
[ 1898.907653] xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
[ 1898.915612] xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
[ 1898.919635] xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x003a
[ 1898.923634] xone-gip gip0: gip_process_pkt_chunked: offset=0x00ae, length=0x003a
[ 1898.927608] xone-gip gip0: gip_process_pkt_chunked: offset=0x00e8, length=0x000c
[ 1898.935596] xone-gip gip0: gip_process_pkt_chunked: offset=0x00f4, length=0x0000
[ 1898.935601] xone-gip gip0: gip_parse_client_commands: command=0x20, length=0x20, options=0x10
[ 1898.935603] xone-gip gip0: gip_parse_client_commands: command=0x09, length=0x09, options=0x08
[ 1898.935605] xone-gip gip0: gip_parse_client_commands: command=0x12, length=0x3c, options=0x10
[ 1898.935607] xone-gip gip0: gip_parse_client_commands: command=0x13, length=0x3c, options=0x08
[ 1898.935608] xone-gip gip0: gip_parse_firmware_versions: version=1.7
[ 1898.935611] xone-gip gip0: gip_parse_capabilities: out=01-02-03-04-06-07
[ 1898.935612] xone-gip gip0: gip_parse_capabilities: in=01-04-05-06-0a
[ 1898.935614] xone-gip gip0: gip_parse_classes: class=Windows.Xbox.Input.Gamepad
[ 1898.935616] xone-gip gip0: gip_parse_interfaces: guid=56ff7697-fd9b-8145-ad45-b645bba526d6
[ 1898.935617] xone-gip gip0: gip_parse_interfaces: guid=2c402e08-df07-e145-a5ab-a3127af197b5
[ 1898.935619] xone-gip gip0: gip_parse_interfaces: guid=e71ff3b8-8673-e940-a9f8-2f21263acfb7
[ 1898.935620] xone-gip gip0: gip_parse_interfaces: guid=fed2ddec-87d3-9442-bd96-1a712e3dc77d
[ 1898.935831] input: Microsoft Xbox Controller as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/gip0/gip0.0/input/input19
[ 1898.935934] xone-gip-gamepad gip0.0: gip_register_client: registered
[ 1899.023719] xone-gip gip0: gip_init_chunk_buffer: command=0x06, length=0x005a
[ 1899.023728] xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
[ 1899.031679] xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x0020
[ 1899.039640] xone-gip gip0: gip_process_pkt_chunked: offset=0x005a, length=0x0000
[ 1899.151661] xone-gip gip0: gip_init_chunk_buffer: command=0x06, length=0x0339
[ 1899.151691] xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
[ 1899.159655] xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
[ 1899.163655] xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x003a
[ 1899.183652] xone-gip gip0: gip_process_pkt_chunked: offset=0x00ae, length=0x003a
[ 1899.187661] xone-gip gip0: gip_process_pkt_chunked: offset=0x00e8, length=0x003a
[ 1899.203651] xone-gip gip0: gip_process_pkt_chunked: offset=0x0122, length=0x003a
[ 1899.207725] xone-gip gip0: gip_process_pkt_chunked: offset=0x015c, length=0x003a
[ 1899.211655] xone-gip gip0: gip_process_pkt_chunked: offset=0x0196, length=0x003a
[ 1899.215651] xone-gip gip0: gip_process_pkt_chunked: offset=0x01d0, length=0x003a
[ 1899.219751] xone-gip gip0: gip_process_pkt_chunked: offset=0x020a, length=0x003a
[ 1899.227652] xone-gip gip0: gip_process_pkt_chunked: offset=0x0244, length=0x003a
[ 1899.231654] xone-gip gip0: gip_process_pkt_chunked: offset=0x027e, length=0x003a
[ 1899.235653] xone-gip gip0: gip_process_pkt_chunked: offset=0x02b8, length=0x003a
[ 1899.239655] xone-gip gip0: gip_process_pkt_chunked: offset=0x02f2, length=0x003a
[ 1899.243631] xone-gip gip0: gip_process_pkt_chunked: offset=0x032c, length=0x000d
[ 1899.251618] xone-gip gip0: gip_process_pkt_chunked: offset=0x0339, length=0x0000
[ 1899.265759] xone-gip-gamepad gip0.0: gip_auth_exchange_rsa: encrypt RSA failed: -17

```

BloomyInDev, 2025-02-16:
Its work on kernel lts (kernel `6.12.13-1-lts`) so maybe something broke in 6.13.

PS: for those that will ask, i updated my machine that was running 6.12, updated it to 6.13, seen that it broke, installed `linux-lts` (so kernel 6.12.13) and it works

Slushnas, 2025-02-16:
@justinlicitis I have the same controller and had the same issue. The controller was recognized but none of the inputs would work.  Here is what I did as a workaround (the xone-wired driver would not work and I had to fall back to the xpad driver):

- Create a new udev rule file at `/etc/udev/rules.d/99-8bitdo-xinput.rules` with the following contents:
```
ACTION=="add", ATTRS{idVendor}=="2dc8", ATTRS{idProduct}=="2016", MODE="0666", DRIVER!="xpad", RUN+="/bin/sh -c 'modprobe xpad; echo $kernel > /sys/bus/usb/drivers/xone-wired/unbind; echo $kernel > /sys/bus/usb/drivers/xpad/bind'"
```
- Ran `sudo udevadm control --reload`
- Unplug and plug the controller back in

Note that installing the xone driver disables loading of the xpad driver so I added the modprobe line here to dynamically load it when the controller is plugged in. Just be aware that if the controller is plugged into the computer on startup that it will not power up automatically with this method - you will need to unplug and plug it back in.

Interestingly an older controller model of the 8BitDo Ultimate Controller for Xbox (the one with the USB cable that is not removable) with product ID 2003 did work without this workaround.

ASCI-Blue, 2025-02-21:
Had kernel 6.12.4-zen, upgraded to 6.13.3-zen and Xbox Wireless Headset no longer works wired. Tried both AUR and manual install. Headset shuts off after about a minute 30 seconds, and there is no handshake noise. Xbox Series controller still works properly.

devloloper, 2025-02-27:
Having the same problem with a Razer Wolverine Ultimate on 6.13.4-zen.

@Slushnas workaround worked for me too after changing vendor and product id.

NixieQueen, 2025-03-04:
I may have found a workaround!
I narrowed the error down to this line in crypto.c:
`tfm = crypto_alloc_akcipher("pkcs1pad(rsa,sha256)", 0, 0);`

I found that removing 'sha256' and the comma fixes this issue for me and I am able to use the controller.

`tfm = crypto_alloc_akcipher("pkcs1pad(rsa)", 0, 0);`

Sorry if this is the wrong place, I am new to github.

justinlicitis, 2025-03-05:
@NixieQueen 

This one worked for me too. Thanks!

RphFede, 2025-03-05:
@NixieQueen Hey, which is the file that I have to edit? Also, I'm actually trying to install drivers but something goes wrong when installing firmware in Linux Kernel.

Slushnas, 2025-03-05:
@RphFede https://github.com/dlundqvist/xone/blob/6b9d59aed71f6de543c481c33df4705d4a590a31/auth/crypto.c#L103

onestan, 2025-03-07:
> I may have found a workaround! I narrowed the error down to this line in crypto.c: `tfm = crypto_alloc_akcipher("pkcs1pad(rsa,sha256)", 0, 0);`
>
> I found that removing 'sha256' and the comma fixes [...]

This worked! Thank you!

dlundqvist, 2025-03-12:
This is fixed now that #37 is merged. Thanks for your patience.

closed 2025-03-12 by dlundqvist

Tweekism, 2025-03-13:
@dlundqvist Thanks mate!

commit 5d1a468a (2025-03-12): gip_send_pkt: Only copy data if non-NULL.

This will, currently, never happen because gip_send_pkt is only called
with NULL data when hdr->packet_length <= GIP_PKT_MAX_LENGTH. It's a
bit weird but gip_send_pkt_simple does it the same way. Not my
prefered solution but works fine for now.

Fixes #33.