dlundqvist/xone #34: Nacon Evol-x Pro controller support

mguijarr, 2025-02-16:
First of all let me thank Xone developers for this wonderful driver.

I would like to help integrating the Nacon Evol-x pro controller.

Indeed, at the moment this is what I get (as posted in Discord forum): everything seems to work fine except X and Y are inverted and
RT, LT do not work. Test with evtest, shows that actioning RT makes ABS_RZ and LT makes ABS_Z.

Does anyone know why I have this ?

Any advice for where to start looking at this problem would be nice.

I looked in gamepad.c, at the moment I do not understand how LT and RT are handled. For X and Y I see BIT(6) and BIT(7), but it
is more complicated than just inverting those two (I would to improve the driver for everybody not having my own special weird version !). I looked at Xbox controller specs too... I am new to this and still lost.

Slushnas, 2025-03-11:
What do you mean by the triggers not working? I think it is expected behavior for the triggers to map to ABS_RZ and ABS_Z. My 8bitdo controllers show the same values in evtest output and I don't have any issues with the triggers in games.

As for X and Y being inverted are you just comparing the output in evtest to the mapped buttons? I do find it odd that X maps to BTN_NORTH instead of BTN_WEST and Y maps to BTN_WEST instead of BTN_NORTH however this seems to be expected behavior. When performing a button test in Steam they are mapped correctly.

If you really wanted to flip the behavior of the X and Y buttons you could edit the `gip_gamepad_op_input` function in `gamepad.c` to check for the vendor and product ID of you controller like this and swap them:
```
static int gip_gamepad_op_input(struct gip_client *client, void *data, u32 len)
{
        ...
        input_report_key(dev, BTN_A, buttons & GIP_GP_BTN_A);
	input_report_key(dev, BTN_B, buttons & GIP_GP_BTN_B);
    if (dev->id.vendor == 0x2dc8 && dev->id.product == 0x2016) {
        // swap X and Y
      	input_report_key(dev, BTN_X, buttons & GIP_GP_BTN_Y);
	input_report_key(dev, BTN_Y, buttons & GIP_GP_BTN_X);
    } else {
      	input_report_key(dev, BTN_X, buttons & GIP_GP_BTN_X);
	input_report_key(dev, BTN_Y, buttons & GIP_GP_BTN_Y);
    }
```

mguijarr, 2025-03-14:
> What do you mean by the triggers not working? I think it is expected behavior for the triggers to map to ABS_RZ and ABS_Z.

Ok I didn't know that... At least in Fortnite it does nothing when pressing triggers with this controller.

> I do find it odd that X maps to BTN_NORTH instead of BTN_WEST and Y maps to BTN_WEST instead of BTN_NORTH however this seems to be expected behavior.

Ok, really ?

@Slushnas thanks for your input

Now my son is using the controller on a Windows PC, so... it works. In turn I use now a PS4 controller, works out of the box. So, I don't know when I will be back to this, sorry

Tweekism, 2025-03-18:
> > I do find it odd that X maps to BTN_NORTH instead of BTN_WEST and Y maps to BTN_WEST instead of BTN_NORTH however this seems to be expected behavior.
> 
> Ok, really ?

Yeah X and Y are mapped backwards in the Linux kernel, and then user applications like steam map them backwards again in order to flip them back, it is a hold over from long long again in the before times.

If they were to fix it in the kernel now it would flip every application everywhere and annoy many people, so they just leave it.

Lawstorant, 2025-07-01:
@mguijarr any updates on this? Still issues with triggers?

closed 2025-07-05 by Lawstorant

mguijarr, 2025-07-08:
I have no time to work on it unfortunately. It still does not work