dlundqvist/xone #36: Controller Unresponsive on Wake from Suspend (Bazzite)

Karatakos, 2025-03-03:
After waking the system from sleep the controller successfully connects to the dongle but is unresponsive in Steam. Since usbreset is not included in the Bazzite image it's not a feasible workaround for Bazzite. To fix, the dongle needs to be unplugged but only after turning the controller off first. That or a system reboot. [Full dmesg log here](https://github.com/user-attachments/files/19054183/dmesg.log), [Current boot log here](https://github.com/user-attachments/files/19054203/boot.log).

Hardware/System: 
1. Xbox Elite 2 (Latest Firmware) connected via the Xbox Wireless Adapter (Unknown Firmware).
2. Bazzite 41 (FROM Fedora Kinoite); Linux 6.13.5-100.bazzite.fc41.x86_64

Repro Steps:
1.  In Steam game mode via the controller put the system to sleep
2. Wake the system from sleep via the Xbox button (back into Steam game mode)

MochaThunder, 2025-03-12:
I am having the same issue on Bazzite and have not found a solution

Randy28CR, 2025-03-12:
Same here. I think it was not like that before.
It would be cool if anyone can address this, please!

dlundqvist, 2025-03-12:
Does it only happen in Steam or for all applications?

Randy28CR, 2025-03-12:
> Does it only happen in Steam or for all applications?

In my case, it happens in every single escenario.
I use Bazzite and they claim they are up to date with this driver in their kernel.
In my case, this is what happens:
I wake up the PC from sleep using my Xbox controller  and dongle in Bazzite (Steam OS game mode), and there is big chance that the controller connects with the dongle but it is just unresponsive.
I can tell that easily because I have set the bright of my Xbox controller to 50% via Konsole command, so whenever I wake up my PC and the brightness is full, it means the controller will be unresponsive. I have to stand up, disconnect the dongle, connect it to a new USB port, and then it works as expected.

Notes:
1. This happens regardless the brightness command I put or not.
2. In my personal situation, it happens mostly when I put my PC to sleep and I leave it for 30mins or more.
I will try to replicate it and I will make a video.

Hopefully this can be addressed. To be honest, this driver is awesome! But that is something really hits HTPC experience

Randy28CR, 2025-03-12:
Here you have a video showing up the problem!
https://drive.google.com/file/d/1-3Zv3C9Pwbq4a0xYydi1Eg5xlz5-2C0x/view?usp=sharing

More notes:

1. My PC has 3 USB A black slots, 2 USB A yellow slots, 1 USB A blue slot and 1 USB C slot.
Curiously, in Bazzite, this error happens less often when the controller is connected to the blue slot or USB C slot.
It happens 90% of the time in the other ones, including the front panel ports.

My brother uses V1 Xbox wireless adapter and in their PC with Linux mint in desktop mode it happens always.

Hope you find this useful and you can fix this issue. I deeply appreciate your great work!!

Karatakos, 2025-03-14:
Just a note that the Bazzite fork is 7 commits behind at the time of writing — commit 6b9d59aed71f6de543c481c33df4705d4a590a31. Though none of those commits seem like they would address the issue, If I get chance I’ll try to run the latest driver build next week. Didn’t try a different USB port admittedly but I’ll also try to switch to a USB 2.0 port.

ZureTz, 2025-03-18:
I'm using Xbox dongle V2 and Xbox One Controller. Same issue here. Got this issue for a few months. This almost always happens when wake up or boot up, after pressing Xbox button the controller is connected but none of the applications recognizes the controller input, and I have to replug the dongle to get it working.

Randy28CR, 2025-03-18:
> I'm using Xbox dongle V2 and Xbox One Controller. Same issue here. Got this issue for a few months. This almost always happens when wake up or boot up, after pressing Xbox button the controller is [...]

I deeply suggest you to plug it in USB blue ports.

ZureTz, 2025-03-18:
> > I'm using Xbox dongle V2 and Xbox One Controller. Same issue here. Got this issue for a few months. This almost always happens when wake up or boot up, after pressing Xbox button the controller is connected but none of the applications recognizes the controller input, 和 I have to replug the dongle to get it working.
> 
> I deeply suggest you to plug it in USB blue ports.

Yes I'm exactly using USB blue ports.

Karatakos, 2025-03-18:
Shouldn’t speak too soon but the Bazzite fork was updated on Saturday pulling in the missing commits including an auth fix and after upgrading to today’s Bazzite image the issue seems to be resolved! Please update to the latest image and try again.

Randy28CR, 2025-03-26:
This has not been addressed yet.
Any clue guys?

karatakos, 2025-03-27:
> This has not been addressed yet. Any clue guys?

An update is literally right above your post. Did you try a Bazzite update??

Randy28CR, 2025-04-07:
This issue is still occuring. Nothing has changed. I have Bazzite up to date and every single time I leave my PC suspended for lets say 30 minutes or 1 hour, when I wake it up the controller does not connect whatsoever 

> Does it only happen in Steam or for all applications?

Please let me know if you need anything from me in order to address this issue. I am willing to donate if needed!

mrgnex, 2025-06-18:
Just experienced this issue today. I was playing a game and the PC went to sleep while I was away. After waking it, the controller wouldn't be recognized anymore although the Xbox logo was lit like it was connected. Can confirm unplugging and plugging the adapter in again fixes it. I switched it to a blue (USB 3.0) port now too. Let's see.

Lawstorant, 2025-06-20:
@mrgnex could you try with #70?

Lawstorant, 2025-08-03:
@karatakos @mrgnex @Randy28CR @ZureTz does it still happen, especially with latest versions of xone?

Lawstorant, 2025-08-19:
I added some suspend fixes in 0.4.3 but you'd have to ask Bazzite maintainers to update xone

Randy28CR, 2025-08-19:
> I added some suspend fixes in 0.4.3 but you'd have to ask Bazzite maintainers to update xone

I will provide this news to the Bazzite team. We deeply appreciate yopur hard work, really!