dlundqvist/xone #38: Xbox One Series 2 Elite Controller not pairing

xNimbus-0, 2025-03-14:
I have an Xbox one series 2 elite controller that wont pair when using a dongle, the controller is working and dected while plugged in via usb-c and the dongle is detected and wants to pair but the two arent talking to each other like they should.

dlundqvist, 2025-03-14:
Has it never worked or did it stop working recently?

xNimbus-0, 2025-03-14:
It has never worked, I was talking in discord and was told to do this

stechermichal, 2025-05-07:
I'm having no problems with presumably same setup.  Xbox Elite Series 2 connected to an Xbox Wireless Adapter for Windows. CachyOS, kernel 6.14. 

If there wasn't an issue like getting an error while trying to install xone, I'd check that xone is running `lsmod | grep xone
` and if not, run with `sudo modprobe xone`. Then unplug and replug the dongle, check the dongle is all good and start pairing on the dongle and pairing on the controller with the button next to LB. 

And if that doesn't work, I'd probably look for an issue elsewhere, like the dongle and maybe check if it's working on windows if you have access to that. Update the controller firmware on windows, but that might be a cope.

closed 2025-07-02 by Lawstorant