dlundqvist/xone #42: Unable to sync LS35X Headset with Dongle

Audience9986, 2025-03-25:
I'm attempting to sync my LucidSound LS35X headset with my xbox one dongle, but hitting the sync buttons just leaves both the headset and dongle blinking. 

DKMS Status:
```
kvmfr/0.0.12, 6.13.8-arch1-1, x86_64: installed
xone/0.3.r101.gaeb27e6, 6.13.8-arch1-1, x86_64: installed
```

dmesg output when syncing:

```
[  675.480904] usb 1-5: new high-speed USB device number 6 using xhci_hcd
[  675.695888] usb 1-5: New USB device found, idVendor=045e, idProduct=02fe, bcdDevice= 1.00
[  675.695894] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  675.695895] usb 1-5: Product: XBOX ACC
[  675.695897] usb 1-5: Manufacturer: Microsoft Inc.
[  675.695898] usb 1-5: SerialNumber: 451347
[  675.918520] usb 1-5: reset high-speed USB device number 6 using xhci_hcd
[  682.915892] xone-gip gip0: gip_init_chunk_buffer: already initialized
[  685.477458] xone-gip gip0: gip_init_chunk_buffer: already initialized
[  688.138353] xone-gip gip0: gip_init_chunk_buffer: already initialized
```

I've tried a couple of different usb ports, but same issue each time.

ykpmusicstudio, 2025-04-13:
The gip_init_chunk_buffer message makes me think about an unattended retransmission of a chunk. The buffer is cleared when acknowledge is done.

I make a couple of changes in this function for a completely unrelated issue which fixes some edge cases regarding chunked packets. You could try to apply https://github.com/dlundqvist/xone/pulls/27 and report whether the behavior has changed or not.

Audience9986, 2025-04-13:
> The gip_init_chunk_buffer message makes me think about an unattended retransmission of a chunk. The buffer is cleared when acknowledge is done.
> 
> I made a couple of changes in this function for a completely unrelated issue which fixes some edge cases regarding chunked packets. You could try to apply [#46](https://github.com/dlundqvist/xone/pull/46) and report whether the behavior has changed or not.
> 
> **note:** the PR [#46](https://github.com/dlundqvist/xone/pull/46) was built to fix issues for wired devices, so it's not guaranteed to fix yours, but it may still be worth a try.

Still not working, but seems to have made some progress. Seems to at least register now, but then immidately unregisters. 

``[  448.381855] usb 1-5: reset high-speed USB device number 4 using xhci_hcd
[  523.797885] usb 1-5: reset high-speed USB device number 4 using xhci_hcd
[  599.217892] usb 1-5: reset high-speed USB device number 4 using xhci_hcd
[  629.689998] usb 1-5: USB disconnect, device number 4
[  631.349821] usb 1-5: new high-speed USB device number 5 using xhci_hcd
[  631.563906] usb 1-5: New USB device found, idVendor=045e, idProduct=02fe, bcdDevice= 1.00
[  631.563909] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  631.563911] usb 1-5: Product: XBOX ACC
[  631.563912] usb 1-5: Manufacturer: Microsoft Inc.
[  631.563914] usb 1-5: SerialNumber: 451347
[  631.799418] usb 1-5: reset high-speed USB device number 5 using xhci_hcd
[  632.008998] xone-dongle 1-5:1.0: xone_mt76_send_firmware: build=201703281033____
[  632.125899] xone-dongle 1-5:1.0: xone_mt76_init_radio: id=0x7613
[  632.359912] xone-dongle 1-5:1.0: xone_mt76_init_address: address=62:45:b5:11:89:e8
[  632.395940] xone-dongle 1-5:1.0: xone_mt76_evaluate_channels: channel=1, power=32
[  632.410952] xone-dongle 1-5:1.0: xone_mt76_evaluate_channels: channel=6, power=33
[  632.425952] xone-dongle 1-5:1.0: xone_mt76_evaluate_channels: channel=11, power=35
[  632.440939] xone-dongle 1-5:1.0: xone_mt76_evaluate_channels: channel=36, power=47
[  632.455940] xone-dongle 1-5:1.0: xone_mt76_evaluate_channels: channel=40, power=47
[  632.470952] xone-dongle 1-5:1.0: xone_mt76_evaluate_channels: channel=44, power=48
[  632.485938] xone-dongle 1-5:1.0: xone_mt76_evaluate_channels: channel=48, power=48
[  632.500952] xone-dongle 1-5:1.0: xone_mt76_evaluate_channels: channel=149, power=42
[  632.515946] xone-dongle 1-5:1.0: xone_mt76_evaluate_channels: channel=153, power=42
[  632.530952] xone-dongle 1-5:1.0: xone_mt76_evaluate_channels: channel=157, power=43
[  632.545944] xone-dongle 1-5:1.0: xone_mt76_evaluate_channels: channel=161, power=43
[  632.560952] xone-dongle 1-5:1.0: xone_mt76_evaluate_channels: channel=165, power=43
[  632.563925] xone-dongle 1-5:1.0: xone_mt76_init_channels: channel=44
[  633.645990] usbcore: registered new interface driver xone-dongle
[  644.912544] xone-gip gip0: gip_create_adapter: registered
[  644.912690] xone-dongle 1-5:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:84:5c:00:b6
[  645.026927] xone-gip gip0: gip_get_client: initialized client 0
[  645.026931] xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:84:5c:00:b6, vendor=0x2f12, product=0x0023
[  645.026934] xone-gip gip0: gip_handle_pkt_announce: firmware=2.5.0.0, hardware=1537.1.1.1
[  645.034964] xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:84:5c:00:b6, vendor=0x2f12, product=0x0023
[  645.034968] xone-gip gip0: gip_handle_pkt_announce: firmware=2.5.0.0, hardware=1537.1.1.1
[  645.042997] xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x007e
[  645.043004] xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
[  645.051058] xone-gip gip0: gip_init_chunk_buffer: already initialized (in)
[  645.051063] xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x007e
[  645.051069] xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
[  645.067136] xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
[  645.067388] xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x000a
[  645.067392] xone-gip gip0: gip_process_pkt_chunked: last chunked packet
[  645.083224] xone-gip gip0: gip_process_pkt_chunked: offset=0x007e, length=0x0000
[  645.083228] xone-gip gip0: gip_parse_firmware_versions: version=2.5
[  645.083230] xone-gip gip0: gip_parse_audio_formats: formats=09-10-09-09
[  645.083232] xone-gip gip0: gip_parse_capabilities: out=01-02-03-04-06-08-60
[  645.083234] xone-gip gip0: gip_parse_capabilities: in=01-04-05-06-08-60
[  645.083235] xone-gip gip0: gip_parse_classes: class=Windows.Xbox.Input.Headset
[  645.083236] xone-gip gip0: gip_parse_interfaces: guid=56ff7697-fd9b-8145-ad45-b645bba526d6
[  645.083238] xone-gip gip0: gip_parse_interfaces: guid=a3d125bc-4ec2-9249-9dda-ef4f123ef5dc
[  645.083276] xone-gip gip0.0: gip_register_client: registered
[  646.123802] xone-gip-headset gip0.0: gip_headset_config: format=0x09/0x10
[  646.728678] xone-dongle 1-5:1.0: xone_dongle_handle_loss: wcid=1
[  646.728700] xone-dongle 1-5:1.0: xone_dongle_remove_client: wcid=1, address=7e:ed:84:5c:00:b6
[  646.728744] xone-gip gip0: gip_destroy_adapter: unregistered
[  647.472473] xone-gip gip0: gip_create_adapter: registered
[  647.472628] xone-dongle 1-5:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:84:5c:00:b6
[  647.587468] xone-gip gip0: gip_get_client: initialized client 0
[... cut ...]
[  660.449519] xone-gip gip0: gip_parse_interfaces: guid=56ff7697-fd9b-8145-ad45-b645bba526d6
[  660.449521] xone-gip gip0: gip_parse_interfaces: guid=a3d125bc-4ec2-9249-9dda-ef4f123ef5dc
[  660.449549] xone-gip-headset gip0.0: gip_register_client: registered
[  661.483804] xone-gip-headset gip0.0: gip_headset_config: format=0x09/0x10
[  662.100762] xone-dongle 1-5:1.0: xone_dongle_handle_loss: wcid=1
[  662.100786] xone-dongle 1-5:1.0: xone_dongle_remove_client: wcid=1, address=7e:ed:84:5c:00:b6
[  662.100843] xone-gip gip0: gip_destroy_adapter: unregistered
`

ykpmusicstudio, 2025-04-13:
It fails during gip_headset_config, probably at gip_suggest_audio_format().
The handshake is never started.
I wonder if your device expects handshake to be done before configuring the device (which makes some sense) and  disconnects the device when receiving a message from an unauthenticated host.

ykpmusicstudio, 2025-06-22:
@Audience9986 : could you try my last https://github.com/dlundqvist/xone/pull/46 commit ? I don't think it will fix you issue but I added some new information in the logs. Can you send me same logs from the initial connection with this new commit ?

Audience9986, 2025-06-23:
Still not able to sync.

```
[   27.078587] xone-dongle 1-6:1.0: xone_dongle_init: trying to load firmware xow_dongle.bin
[   27.082084] xone-dongle 1-6:1.0: xone_mt76_send_firmware: build=201703281033____
[   27.199081] xone-dongle 1-6:1.0: xone_mt76_init_radio: id=0x7613
[   27.433138] xone-dongle 1-6:1.0: xone_mt76_init_address: fuse_address=62:45:b5:11:89:e8
[   27.469144] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=1, power=32
[   27.484122] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=6, power=33
[   27.499525] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=11, power=35
[   27.514514] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=36, power=47
[   27.529169] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=40, power=47
[   27.544140] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=44, power=48
[   27.559129] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=48, power=48
[   27.574121] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=149, power=42
[   27.589120] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=153, power=42
[   27.604128] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=157, power=43
[   27.619120] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=161, power=43
[   27.634130] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=165, power=43
[   27.637086] xone-dongle 1-6:1.0: xone_mt76_init_channels: channel=44
[   28.197304] br0: port 1(enp8s0) entered forwarding state
[   28.440254] [UFW BLOCK] IN=br0 OUT= MAC= SRC=192.168.1.152 DST=224.0.0.252 LEN=54 TOS=0x00 PREC=0x00 TTL=255 ID=28896 PROTO=UDP SPT=5355 DPT=5355 LEN=34 
[   28.707136] [UFW BLOCK] IN=br0 OUT= MAC= SRC=192.168.1.152 DST=224.0.0.252 LEN=54 TOS=0x00 PREC=0x00 TTL=255 ID=29038 PROTO=UDP SPT=5355 DPT=5355 LEN=34 
[   28.712157] usbcore: registered new interface driver xone-dongle
[   28.960009] [UFW BLOCK] IN=br0 OUT= MAC= SRC=192.168.1.152 DST=224.0.0.252 LEN=54 TOS=0x00 PREC=0x00 TTL=255 ID=29189 PROTO=UDP SPT=5355 DPT=5355 LEN=34 
[   30.161778] [UFW BLOCK] IN=br0 OUT= MAC= SRC=fe80:0000:0000:0000:bac8:ed6b:bad4:d92a DST=ff02:0000:0000:0000:0000:0000:0001:0003 LEN=74 TC=0 HOPLIMIT=255 FLOWLBL=565263 PROTO=UDP SPT=5355 DPT=5355 LEN=34 
[   30.429364] [UFW BLOCK] IN=br0 OUT= MAC= SRC=fe80:0000:0000:0000:bac8:ed6b:bad4:d92a DST=ff02:0000:0000:0000:0000:0000:0001:0003 LEN=74 TC=0 HOPLIMIT=255 FLOWLBL=565263 PROTO=UDP SPT=5355 DPT=5355 LEN=34 
[   30.860162] [UFW BLOCK] IN=br0 OUT= MAC= SRC=fe80:0000:0000:0000:bac8:ed6b:bad4:d92a DST=ff02:0000:0000:0000:0000:0000:0001:0003 LEN=74 TC=0 HOPLIMIT=255 FLOWLBL=565263 PROTO=UDP SPT=5355 DPT=5355 LEN=34 
[   34.961030] xone-gip gip0: gip_create_adapter: registered
[   34.961175] xone-dongle 1-6:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:84:5c:00:b6
[   35.081082] xone-gip gip0: gip_get_client: initialized client 0
[   35.081086] xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:84:5c:00:b6, vendor=0x2f12, product=0x0023
[   35.081089] xone-gip gip0: gip_handle_pkt_announce: firmware=2.5.0.0, hardware=1537.1.1.1
[   35.081338] xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:84:5c:00:b6, vendor=0x2f12, product=0x0023
[   35.081340] xone-gip gip0: gip_handle_pkt_announce: firmware=2.5.0.0, hardware=1537.1.1.1
[   35.097426] xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x007e
[   35.097432] xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
[   35.105485] xone-gip gip0: gip_init_chunk_buffer: already initialized
[   35.105487] xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x007e
[   35.105490] xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
[   35.121546] xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
[   35.121803] xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x000a
[   35.137637] xone-gip gip0: gip_process_pkt_chunked: offset=0x007e, length=0x0000
[   35.137639] xone-gip gip0: gip_parse_firmware_versions: version=2.5
[   35.137641] xone-gip gip0: gip_parse_audio_formats: formats=09-10-09-09
[   35.137642] xone-gip gip0: gip_parse_capabilities: out=01-02-03-04-06-08-60
[   35.137644] xone-gip gip0: gip_parse_capabilities: in=01-04-05-06-08-60
[   35.137645] xone-gip gip0: gip_parse_classes: class=Windows.Xbox.Input.Headset
[   35.137646] xone-gip gip0: gip_parse_interfaces: guid=56ff7697-fd9b-8145-ad45-b645bba526d6
[   35.137648] xone-gip gip0: gip_parse_interfaces: guid=a3d125bc-4ec2-9249-9dda-ef4f123ef5dc
[   35.137677] xone-gip gip0.0: gip_register_client: registered
[   36.197304] xone-gip-headset gip0.0: gip_headset_config: format=0x09/0x10
[   36.910147] xone-dongle 1-6:1.0: xone_dongle_handle_loss: wcid=1
[   36.910155] xone-dongle 1-6:1.0: xone_dongle_remove_client: wcid=1, address=7e:ed:84:5c:00:b6
[   36.910191] xone-gip gip0: gip_destroy_adapter: unregistered
[   37.455520] systemd-journald[393]: Time jumped backwards, rotating.
[   37.623386] xone-gip gip0: gip_create_adapter: registered
[   37.623527] xone-dongle 1-6:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:84:5c:00:b6
[   37.743658] xone-gip gip0: gip_get_client: initialized client 0
[   37.743661] xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:84:5c:00:b6, vendor=0x2f12, product=0x0023
[   37.743663] xone-gip gip0: gip_handle_pkt_announce: firmware=2.5.0.0, hardware=1537.1.1.1
[   37.743915] xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:84:5c:00:b6, vendor=0x2f12, product=0x0023
[   37.743917] xone-gip gip0: gip_handle_pkt_announce: firmware=2.5.0.0, hardware=1537.1.1.1
[   37.760014] xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x007e
[   37.760019] xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
[   37.768343] xone-gip gip0: gip_init_chunk_buffer: already initialized

```

Audience9986, 2025-06-24:
I pulled it again to make sure this time

```
kvmfr/0.0.12, 6.15.3-3-cachyos, x86_64: installed
xone/v0.3.1-13-g20feb3b, 6.15.3-3-cachyos, x86_64: installed

```

```
[  253.992276] usb 1-6: USB disconnect, device number 6
[  255.740464] usb 1-6: new high-speed USB device number 7 using xhci_hcd
[  255.955753] usb 1-6: New USB device found, idVendor=045e, idProduct=02fe, bcdDevice= 1.00
[  255.955756] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  255.955758] usb 1-6: Product: XBOX ACC
[  255.955759] usb 1-6: Manufacturer: Microsoft Inc.
[  255.955760] usb 1-6: SerialNumber: 451347
[  256.180014] usb 1-6: reset high-speed USB device number 7 using xhci_hcd
[  256.424343] xone-dongle 1-6:1.0: xone_dongle_init: trying to load firmware xow_dongle.bin
[  256.426755] xone-dongle 1-6:1.0: xone_mt76_send_firmware: build=201703281033____
[  256.543753] xone-dongle 1-6:1.0: xone_mt76_init_radio: id=0x7613
[  256.585794] xone-dongle 1-6:1.0: xone_dongle_toggle_pairing: enabled=1
[  256.780759] xone-dongle 1-6:1.0: xone_mt76_init_address: fuse_address=62:45:b5:11:89:e8
[  256.816792] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=1, power=32
[  256.831791] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=6, power=33
[  256.846792] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=11, power=35
[  256.861791] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=36, power=47
[  256.876792] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=40, power=47
[  256.891792] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=44, power=48
[  256.906791] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=48, power=48
[  256.921791] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=149, power=42
[  256.936791] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=153, power=42
[  256.951791] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=157, power=43
[  256.966792] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=161, power=43
[  256.981795] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=165, power=43
[  256.984755] xone-dongle 1-6:1.0: xone_mt76_init_channels: channel=44
[  262.886051] xone-gip gip0: gip_create_adapter: registered
[  262.886153] xone-dongle 1-6:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:84:5c:00:b6
[  263.006127] xone-gip gip0: gip_get_client: initialized client 0
[  263.006130] xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:84:5c:00:b6, vendor=0x2f12, product=0x0023
[  263.006133] xone-gip gip0: gip_handle_pkt_announce: firmware=2.5.0.0, hardware=1537.1.1.1
[  263.006375] xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:84:5c:00:b6, vendor=0x2f12, product=0x0023
[  263.006377] xone-gip gip0: gip_handle_pkt_announce: firmware=2.5.0.0, hardware=1537.1.1.1
[  263.022479] xone-gip gip0: gip_init_chunk_buffer[in]: command=0x04, length=0x007e
[  263.022482] xone-gip gip0: gip_process_pkt_chunked: flags=[Ini Ack Sys], offset=0x0000, length=0x003a
[  263.022484] xone-gip gip0: gip_acknowledge_pkt: ACME command=0x04, length=0x003a
[  263.022737] xone-gip gip0: gip_init_chunk_buffer: already initialized (in)
[  263.022739] xone-gip gip0: gip_init_chunk_buffer[in]: command=0x04, length=0x007e
[  263.022741] xone-gip gip0: gip_process_pkt_chunked: flags=[Ini Ack Sys], offset=0x0000, length=0x003a
[  263.022743] xone-gip gip0: gip_acknowledge_pkt: ACME command=0x04, length=0x003a
[  263.038589] xone-gip gip0: gip_process_pkt_chunked: flags=[... ... Sys], offset=0x003a, length=0x003a
[  263.046626] xone-gip gip0: gip_process_pkt_chunked: flags=[... Ack Sys], offset=0x0074, length=0x000a
[  263.046629] xone-gip gip0: gip_acknowledge_pkt: ACME command=0x04, length=0x007e
[  263.062927] xone-gip gip0: gip_process_pkt_chunked: flags=[... ... Sys], offset=0x007e, length=0x0000
[  263.062930] xone-gip gip0: gip_parse_firmware_versions: version=2.5
[  263.062933] xone-gip gip0: gip_parse_audio_formats: formats=09-10-09-09
[  263.062935] xone-gip gip0: gip_parse_capabilities: out=01-02-03-04-06-08-60
[  263.062937] xone-gip gip0: gip_parse_capabilities: in=01-04-05-06-08-60
[  263.062938] xone-gip gip0: gip_parse_classes: class=Windows.Xbox.Input.Headset
[  263.062940] xone-gip gip0: gip_parse_interfaces: guid=56ff7697-fd9b-8145-ad45-b645bba526d6
[  263.062942] xone-gip gip0: gip_parse_interfaces: guid=a3d125bc-4ec2-9249-9dda-ef4f123ef5dc
[  263.062984] xone-gip-headset gip0.0: gip_register_client: registered
[  264.083888] xone-gip-headset gip0.0: gip_headset_config: format=0x09/0x10
[  264.097968] xone-gip gip0: gip_make_audio_config: rate=24000/1, buffer=384
[  264.097971] xone-gip gip0: gip_make_audio_config: rate=48000/2, buffer=1536
```

ykpmusicstudio, 2025-06-25:
The version's right, this the kind of logs I expected to read.
Up to the end of this excerpt the dialog between the driver and the device looks fine, if anything went wrong it happened after the last line.
Does it abruptly stop at time = 264.097971 or is there more xone-gip / xone-dongle / xone-gip-headset logs ?

Audience9986, 2025-06-26:
It doesn't stop, but it will just repeat until I turn the headset off. Here's a longer output of dmesg 

```
[ 5818.230957] usb 1-6: new high-speed USB device number 5 using xhci_hcd
[ 5818.446138] usb 1-6: New USB device found, idVendor=045e, idProduct=02fe, bcdDevice= 1.00
[ 5818.446141] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5818.446143] usb 1-6: Product: XBOX ACC
[ 5818.446144] usb 1-6: Manufacturer: Microsoft Inc.
[ 5818.446146] usb 1-6: SerialNumber: 451347
[ 5818.669545] usb 1-6: reset high-speed USB device number 5 using xhci_hcd
[ 5818.913866] xone-dongle 1-6:1.0: xone_dongle_init: trying to load firmware xow_dongle.bin
[ 5818.916146] xone-dongle 1-6:1.0: xone_mt76_send_firmware: build=201703281033____
[ 5819.034152] xone-dongle 1-6:1.0: xone_mt76_init_radio: id=0x7613
[ 5819.270149] xone-dongle 1-6:1.0: xone_mt76_init_address: fuse_address=62:45:b5:11:89:e8
[ 5819.306184] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=1, power=32
[ 5819.321183] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=6, power=33
[ 5819.336184] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=11, power=35
[ 5819.351210] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=36, power=47
[ 5819.366357] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=40, power=47
[ 5819.381184] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=44, power=48
[ 5819.396185] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=48, power=48
[ 5819.411189] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=149, power=42
[ 5819.426184] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=153, power=42
[ 5819.441184] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=157, power=43
[ 5819.456184] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=161, power=43
[ 5819.471184] xone-dongle 1-6:1.0: xone_mt76_evaluate_channels: channel=165, power=43
[ 5819.474149] xone-dongle 1-6:1.0: xone_mt76_init_channels: channel=44
[ 5820.077191] xone-dongle 1-6:1.0: xone_dongle_toggle_pairing: enabled=1
[ 5831.446147] xone-gip gip0: gip_create_adapter: registered
[ 5831.446250] xone-dongle 1-6:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:84:5c:00:b6
[ 5831.566374] xone-gip gip0: gip_get_client: initialized client 0
[ 5831.566378] xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:84:5c:00:b6, vendor=0x2f12, product=0x0023
[ 5831.566380] xone-gip gip0: gip_handle_pkt_announce: firmware=2.5.0.0, hardware=1537.1.1.1
[ 5831.566620] xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:84:5c:00:b6, vendor=0x2f12, product=0x0023
[ 5831.566622] xone-gip gip0: gip_handle_pkt_announce: firmware=2.5.0.0, hardware=1537.1.1.1
[ 5831.582713] xone-gip gip0: gip_init_chunk_buffer[in]: command=0x04, length=0x007e
[ 5831.582715] xone-gip gip0: gip_process_pkt_chunked: flags=[Ini Ack Sys], offset=0x0000, length=0x003a
[ 5831.582717] xone-gip gip0: gip_acknowledge_pkt: ACME command=0x04, length=0x003a
[ 5831.590977] xone-gip gip0: gip_init_chunk_buffer: already initialized (in)
[ 5831.590979] xone-gip gip0: gip_init_chunk_buffer[in]: command=0x04, length=0x007e
[ 5831.590981] xone-gip gip0: gip_process_pkt_chunked: flags=[Ini Ack Sys], offset=0x0000, length=0x003a
[ 5831.590982] xone-gip gip0: gip_acknowledge_pkt: ACME command=0x04, length=0x003a
[ 5831.607057] xone-gip gip0: gip_process_pkt_chunked: flags=[... ... Sys], offset=0x003a, length=0x003a
[ 5831.607311] xone-gip gip0: gip_process_pkt_chunked: flags=[... Ack Sys], offset=0x0074, length=0x000a
[ 5831.607313] xone-gip gip0: gip_acknowledge_pkt: ACME command=0x04, length=0x007e
[ 5831.623153] xone-gip gip0: gip_process_pkt_chunked: flags=[... ... Sys], offset=0x007e, length=0x0000
[ 5831.623156] xone-gip gip0: gip_parse_firmware_versions: version=2.5
[ 5831.623157] xone-gip gip0: gip_parse_audio_formats: formats=09-10-09-09
[ 5831.623159] xone-gip gip0: gip_parse_capabilities: out=01-02-03-04-06-08-60
[ 5831.623160] xone-gip gip0: gip_parse_capabilities: in=01-04-05-06-08-60
[ 5831.623162] xone-gip gip0: gip_parse_classes: class=Windows.Xbox.Input.Headset
[ 5831.623163] xone-gip gip0: gip_parse_interfaces: guid=56ff7697-fd9b-8145-ad45-b645bba526d6
[ 5831.623164] xone-gip gip0: gip_parse_interfaces: guid=a3d125bc-4ec2-9249-9dda-ef4f123ef5dc
[ 5831.623198] xone-gip gip0.0: gip_register_client: registered
[ 5832.695190] xone-gip-headset gip0.0: gip_headset_config: format=0x09/0x10
[ 5832.711757] xone-gip gip0: gip_make_audio_config: rate=24000/1, buffer=384
[ 5832.711760] xone-gip gip0: gip_make_audio_config: rate=48000/2, buffer=1536
[ 5833.272862] xone-dongle 1-6:1.0: xone_dongle_handle_loss: wcid=1
[ 5833.272870] xone-dongle 1-6:1.0: xone_dongle_remove_client: wcid=1, address=7e:ed:84:5c:00:b6
[ 5833.272918] xone-gip gip0: gip_destroy_adapter: unregistered
[ 5834.006292] xone-gip gip0: gip_create_adapter: registered
[ 5834.006397] xone-dongle 1-6:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:84:5c:00:b6
[ 5834.126621] xone-gip gip0: gip_get_client: initialized client 0
[ 5834.126624] xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:84:5c:00:b6, vendor=0x2f12, product=0x0023
[ 5834.126627] xone-gip gip0: gip_handle_pkt_announce: firmware=2.5.0.0, hardware=1537.1.1.1
[ 5834.126893] xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:84:5c:00:b6, vendor=0x2f12, product=0x0023
[ 5834.126896] xone-gip gip0: gip_handle_pkt_announce: firmware=2.5.0.0, hardware=1537.1.1.1
[... cut ...]
[ 5859.169484] xone-gip gip0: gip_parse_capabilities: out=01-02-03-04-06-08-60
[ 5859.169486] xone-gip gip0: gip_parse_capabilities: in=01-04-05-06-08-60
[ 5859.169487] xone-gip gip0: gip_parse_classes: class=Windows.Xbox.Input.Headset
[ 5859.169488] xone-gip gip0: gip_parse_interfaces: guid=56ff7697-fd9b-8145-ad45-b645bba526d6
[ 5859.169490] xone-gip gip0: gip_parse_interfaces: guid=a3d125bc-4ec2-9249-9dda-ef4f123ef5dc
[ 5859.169535] xone-gip-headset gip0.0: gip_register_client: registered
```

ykpmusicstudio, 2025-06-26:
Ok, makes sense to cut where it repeats.

This behavior is quite different from others headset that were failing at AUTHENTICATE, this headset disconnects just before the driver has a chance to start handshake. PR https://github.com/dlundqvist/xone/pull/46 is therefore not significant here.

Nonetheless, it is quite consistent in its behavior with or without my PR so it's definitely related to something the driver has always been doing. It happens between headset config and register / start device.

I know that headset.c is not completely in line with spec in its behavior, the LS35X device may have a strict implementation.

I suspect a timeout issue or a sequence that is not compliant with what the headset expect during audio format exchange. This could be a game of trial and error I think. Starting by introducing some new logs to catch the exchanges with the headset just prior to its disconnection.

I'll try to propose a PR to do that, that I'll link in this issue when ready.

ykpmusicstudio, 2025-06-29:
I created a new PR (https://github.com/dlundqvist/xone/pull/75) to advance in fixing this issue. At this point in time no changes are made to the driver but adding a couple of logs to figure out what happens before the headset disconnects.

Can you try it and send the logs, please.

I you prefer, any further tests and discussion can be made on the discord channel, I opened a thread (https://discord.com/channels/733964971842732042/1388912049672814682).

ykpmusicstudio, 2025-08-09:
@Audience9986 A probable fix is released in https://github.com/dlundqvist/xone/pull/75. Fingers crossed.

ykpmusicstudio, 2026-08-31:
@Audience9986  Please try #200 which overhauls the power on and auth exchanges.

Audience9986, 2026-11-16:
I'm sorry I've not responded, but I've since stopped using the LS35X and had packed it away. I wanted to see if Claude could figure this out so I pulled it out of storage. After some trial and error, I do have audio for it. Hope this helps. I can also send the pcapng files if needed that Claude had me record. 

[message-for-developer.md](https://github.com/user-attachments/files/34929437/message-for-developer.md)
[ls35x-fix.patch](https://github.com/user-attachments/files/34929441/ls35x-fix.patch)