dlundqvist/xone #48: 8bitdo wireless adapter fails to connect

forkymcforkface, 2025-04-14:
The 8bitdo wireless adapter V2 looks to change its device type during the connection process with and ultimately fails after the baudrate tries to set in with the xone driver.

```
[  158.440521] usb 1-2: new full-speed USB device number 81 using xhci-hcd
[  158.613852] usb 1-2: New USB device found, idVendor=2dc8, idProduct=3106, bcdDevice= 1.00
[  158.613858] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  158.613860] usb 1-2: Product: 8BitDo Receiver
[  158.613861] usb 1-2: Manufacturer: 8BitDo
[  158.613863] usb 1-2: SerialNumber: E417D86B1984
[  160.155425] usb 1-2: USB disconnect, device number 81
[  160.548469] usb 1-2: new full-speed USB device number 82 using xhci-hcd
[  160.723972] usb 1-2: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.14
[  160.723978] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  160.723980] usb 1-2: Product: Controller
[  160.723982] usb 1-2: Manufacturer: 8BitDo
[  160.723983] usb 1-2: SerialNumber: E417D86B1984
[  162.240754] usb 1-2: USB disconnect, device number 82
[  162.636403] usb 1-2: new full-speed USB device number 83 using xhci-hcd
[  162.810089] usb 1-2: New USB device found, idVendor=057e, idProduct=2009, bcdDevice= 2.10
[  162.810094] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  162.810096] usb 1-2: Product: Pro Controller
[  162.810097] usb 1-2: Manufacturer: Nintendo Co., Ltd.
[  162.810099] usb 1-2: SerialNumber: E417D86B1984
[  162.867143] nintendo 0003:057E:2009.001D: hidraw0: USB HID v81.11 Joystick [Nintendo Co., Ltd. Pro Controller] on usb-xhci-hcd.0-2/input0
[  163.943236] nintendo 0003:057E:2009.001D: Failed to set baudrate; ret=-71
[  163.943245] nintendo 0003:057E:2009.001D: Failed to initialize controller; ret=-71
[  163.943366] nintendo 0003:057E:2009.001D: probe - fail = -71
[  163.943379] nintendo: probe of 0003:057E:2009.001D failed with error -71
[  163.943494] usb 1-2: USB disconnect, device number 83
```

Here is how it properly connects wtih xpad

```
[  148.676842] usb 1-2: new full-speed USB device number 15 using xhci-hcd
[  148.850441] usb 1-2: New USB device found, idVendor=2dc8, idProduct=3106, bcdDevice= 1.00
[  148.850449] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  148.850451] usb 1-2: Product: 8BitDo Receiver
[  148.850452] usb 1-2: Manufacturer: 8BitDo
[  148.850454] usb 1-2: SerialNumber: E417D86B1984
[  150.368215] usb 1-2: USB disconnect, device number 15
[  150.749044] usb 1-2: new full-speed USB device number 16 using xhci-hcd
[  150.928560] usb 1-2: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.14
[  150.928567] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  150.928569] usb 1-2: Product: Controller
[  150.928571] usb 1-2: Manufacturer: 8BitDo
[  150.928572] usb 1-2: SerialNumber: E417D86B1984
[  152.453461] usb 1-2: USB disconnect, device number 16
[  152.845244] usb 1-2: new full-speed USB device number 17 using xhci-hcd
[  153.018680] usb 1-2: New USB device found, idVendor=057e, idProduct=2009, bcdDevice= 2.10
[  153.018688] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  153.018690] usb 1-2: Product: Pro Controller
[  153.018691] usb 1-2: Manufacturer: Nintendo Co., Ltd.
[  153.018693] usb 1-2: SerialNumber: E417D86B1984
[  153.066753] input: Nintendo Co., Ltd. Pro Controller as /devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-2/1-2:1.0/0003:057E:2009.000A/input/input11
[  153.066890] hid-generic 0003:057E:2009.000A: input,hidraw2: USB HID v1.11 Joystick [Nintendo Co., Ltd. Pro Controller] on usb-xhci-hcd.0-2/input0
```

ykpmusicstudio, 2025-05-24:
Reminds me of a similar issue on MiSTer project (https://github.com/MiSTer-devel/Main_MiSTer/issues/297 and https://misterfpga.org/viewtopic.php?t=1165) or even in the chromium project (https://issues.chromium.org/issues/40933399), that brought up a couple of issues with the 8bitdo dongle. You may find some hints / solutions there, maybe it is related to the various compatibility modes that the 8BitDo has.

From the logs, it doesn't seem clear if xone module is even loaded in the probing phase as the issue happens quite early.

forkymcforkface, 2025-05-24:
I ended up forking xpad and xone and "fixed" it by modifying both so they manage separate devices in harmony. Xone for official Xbox one devices only and xpad for everything else. It works great!

https://github.com/forkymcforkface/xone
https://github.com/forkymcforkface/xpad-noone-ms

> Reminds me of a similar issue on MiSTer project (https://github.com/MiSTer-devel/Main_MiSTer/issues/297 and https://misterfpga.org/viewtopic.php?t=1165) or even in the chromium project (https://issues.chromium.org/issues/40933399), that brought up a couple of issues with the 8bitdo dongle. Some pointed the potential need firmware updates as well (https://www.omgubuntu.co.uk/2019/06/8bitdo-sn30-pro-controller-linux-firmware). You may find some hints / solutions in those links, it may also be related to the various compatibility modes that the 8BitDo has.
> 
> From the logs, it doesn't seem clear if xone module is even loaded in the probing phase as the issue happens quite early.

VladimirMrzv, 2025-06-08:
@forkymcforkface  please update modules to new 6.15 kernel and correct path in instructions

Lawstorant, 2025-07-01:
I don't actually see anything related to `xone` here.

closed 2025-07-01 by Lawstorant