dlundqvist/xone #51: Any plans for a proper release/tag?

tgurr, 2025-04-16:
Comming from a distribution packaging point of view it would be really helpful to have proper new releases to consider switching over our xone package from https://github.com/medusalix/xone to this active fork and I guess other distributions may also consider this as a requirement. Are there any plans yet for a proper new release/tag?

dlundqvist, 2025-04-17:
Yes, this is something that I've started to consider. I'll try to reach out to medusalix first to make sure he's OK with it (I know this is OSS but I still want do this right by him).

medusalix, 2025-04-18:
Hi @dlundqvist, thanks for reaching out! To be honest, I'd really like to keep the current versioning scheme for the original project and I think it would be kind of confusing to release different versions of `xone` from two repositories.
My current plan is to do a final `v0.4` release for kernel versions down to 5.13. After that, I'd like to remove all the compatibility code to focus on the mainlining efforts for `xone`.

If you really want to go ahead with the release process for this repository, I'd maybe add a suffix to the tag to differentiate between the two projects.

dlundqvist, 2025-04-19:
@medusalix, of course if you have plans to wrap up out-of-tree version with v0.4 please do that. Do you have any idea of time frame? Also, please take a look at the changes done in this repo to not miss out on changes and fixes being made.

If you need/want help with mainlining just say so.

wantija, 2025-04-22:
@medusalix Don't want too sound like a jerk since I know life happens, but you won't disappear for almost a year again right? And why not open an issues tracker on your repo so people don't have to go through Discord?

Tweekism, 2025-04-23:
...And he's gone 🤣🤣🤣

Tehe, but speaking of mainlining, I wonder if that discussion is going on elsewhere and that is why the main "medusalix" repo is so inactive.

Lawstorant, 2025-04-23:
@wantija I have the same feeling. It's of course okay if he wants to continue in his own repository but community started to move on finally. Many packaging maintainers already moved to this one or are in the process.

Most importantly, I don't think discouraging other forks from making releases is a good thing especially as 6.15 will break the non-git packages again in a few weeks. AUR package maintainer is waiting for a tag to change over to this repo.

whatever the way will be, we need the tag/release now and medusalix can just recreate the same release on the same commit if he wants to merge this fork's main branch.

wantija, 2025-04-26:
Would a release like 3.1 work for this repo? So that @medusalix can do the 4.0 when he has time and so that this could potentially be packaged in other distributions?

Lawstorant, 2025-04-26:
Honestly, it doesn't matter. The last tag is from 2022...

tgurr, 2025-05-21:
@dlundqvist Huge thanks for tagging 0.3.1 (https://github.com/dlundqvist/xone/releases/tag/v0.3.1) greatly appreciated and also thanks to everyone contributing to this project and keeping things alive and of course also to the origin project and @medusalix!

closed 2025-05-21 by tgurr