dlundqvist/xone #52: Have to unplug and replug the adaptor after every update

Mitchell-Lewis, 2025-04-27:
After every update and dkms module rebuild, i have to unplug and re-plug the adaptor to get the controller to connect.

Lawstorant, 2025-04-28:
Yeah, that's unfortunately how drivers work on Linux.

Lawstorant, 2025-04-30:
That's not really a driver issue though. What you could do is to create a hook for your package manager that maybe could unload the module and load it again after update.

I'll try doing something like that with Arch Linux.

mrgnex, 2025-06-18:
> That's not really a driver issue though. What you could do is to create a hook for your package manager that maybe could unload the module and load it again after update.
> 
> I'll try doing something like that with Arch Linux.

Did you manage to? I am having the same issue.

Lawstorant, 2025-06-25:
@mrgnex your issue is probably the same as #36 or #61

closed 2025-06-25 by Lawstorant