dlundqvist/xone #55: Dongle plugged in at boot creates an immortal udev worker

phesch, 2025-05-09:
I've been encountering an issue using the xone driver on my desktop that I haven't seen mentioned here or in the issues before:
I use the wireless controller with the dongle. If the dongle is plugged in when I boot the PC, I sometimes get the following dmesg entries:
```
usb 3-1: reset high-speed USB device number 2 using xhci_hcd
usb 3-1: can't restore configuration #1 (error=-71)
usb 3-1: USB disconnect, device number 2
```
and an immortal udev worker that cannot be killed (not even with a root SIGKILL) and that occupies an entire CPU core constantly (as if it was busy-waiting or something). The udev worker also appears to block any other udev-related operations, which leads to Fun(TM) if I forget and try to run a system upgrade while it's running. If I plug the dongle in after boot or the bug happens to not trigger (it's inconsistent, but I don't know what it depends on), I get this dmesg output instead and everything works:
```
usb 3-1: new high-speed USB device number 3 using xhci_hcd
usb 3-1: New USB device found, idVendor=045e, idProduct=02e6, bcdDevice= 1.00
usb 3-1: New USB device string: Mfr=1, Product=2, SerialNumber=3
usb 3-1: Product: XBOX ACC
usb 3-1: Manufacturer: Microsoft Inc.
usb 3-1: SerialNumber: 351678
usb 3-1: reset high-speed USB device number 3 using xhci_hcd
usbcore: registered new interface driver xone-dongle
```
Version info:
- Arch Linux
- Kernel 6.14.4-arch1-2
- xone-dkms-git 0.3.r116.g197b160-1
- xone-dongle-firmware 1.0.46.1-2

phesch, 2025-05-09:
I can also confirm that the udev worker in question is the one responsible for the dongle device. Listing the process's open files using lsof shows that it is the only udev worker that opens the `/usr/lib/modules/6.14.4-arch1-2/updates/dkms/xone-{gip,dongle}.ko.zst` files.

phesch, 2025-05-09:
The error shown in the dmesg output is thrown from `usb_reset_and_verify_device` in `drivers/usb/core/hub.c` in the kernel in the following code:
```c
	ret = usb_control_msg(udev, usb_sndctrlpipe(udev, 0),
			USB_REQ_SET_CONFIGURATION, 0,
			udev->actconfig->desc.bConfigurationValue, 0,
			NULL, 0, USB_CTRL_SET_TIMEOUT);
	if (ret < 0) {
		dev_err(&udev->dev,
			"can't restore configuration #%d (error=%d)\n",
			udev->actconfig->desc.bConfigurationValue, ret);
		mutex_unlock(hcd->bandwidth_mutex);
		goto re_enumerate;
	}
```
The error value -71 corresponds to a protocol error `EPROTO`. I don't know why this error is thrown or why it would cause the dongle's udev worker to spin forever.

phesch, 2025-07-01:
In the meantime, I have upgraded to:
- Kernel 6.14.7-arch2-1
- xone-dkms-git 0.3.1.r7.gb8f90da-1

...and I can confirm that even after multiple attempts to provoke it, the issue doesn't happen anymore.
My favorite kind of problem is the kind that fixes itself. I'll close this issue now - thanks for looking into it!

Lawstorant, 2025-08-19:
Thanks! I just added even more fixes around device resets so I'm hopeful this could fix some additional issues like this one.