dlundqvist/xone #57: Temporary input freeze

Nodens-, 2025-05-24:
Ok so here's this strange issue.

Fedora 42, currently on 6.14.7-301.fc42.x86_64 but this has been happening with every kernel tested up to now.
xone package from the fedora copr which is pulling from this repo.
Controller is a Steelseries Xbox One (Cyberpunk 2077 edition) with firmware updated to latest in Windows VM and the lpf firmware package loaded.

The issue I am seeing has these symptoms:
1. If steam is running and I turn on the controller (connected via dongle), mouse and keyboard input freezes for ~10-15secs then resumes normally and everything is working fine. The same thing happens if I turn off the controller.
3. If steam is running and I connect the controller via usb (wired), again same issue as 1.
4. If steam is not running turning the controller on when connected via dongle or connecting it via usb, does not produce the issue.
5. If steam is not running, the controller is on and I start steam, I get the same issue while steam is starting up.
6. If steam is not running, the controller is connected via usb and I start steam, again I get the issue.

Input devices present on the workstation:
```
[#]> ls -l /dev/input/by-id
total 0
lrwxrwxrwx. 1 root root 10 May 24 13:20 usb-Microsoft_Controller_3039455230343931353637303038-event-joystick -> ../event23
lrwxrwxrwx. 1 root root  6 May 24 13:20 usb-Microsoft_Controller_3039455230343931353637303038-joystick -> ../js0
lrwxrwxrwx. 1 root root  9 May 22 10:30 usb-SteelSeries_SteelSeries_Apex_Pro-event-if03 -> ../event5
lrwxrwxrwx. 1 root root  9 May 22 10:30 usb-SteelSeries_SteelSeries_Apex_Pro-event-kbd -> ../event3
lrwxrwxrwx. 1 root root  9 May 22 10:30 usb-SteelSeries_SteelSeries_Apex_Pro-if02-event-kbd -> ../event4
lrwxrwxrwx. 1 root root  9 May 22 10:30 usb-SteelSeries_SteelSeries_Apex_Pro-if03-event-mouse -> ../event6
lrwxrwxrwx. 1 root root  9 May 22 10:30 usb-SteelSeries_SteelSeries_Apex_Pro-if03-mouse -> ../mouse0
lrwxrwxrwx. 1 root root  9 May 22 10:30 usb-SteelSeries_SteelSeries_QCK_Prism-event-if01 -> ../event7
lrwxrwxrwx. 1 root root  9 May 22 10:30 usb-SteelSeries_SteelSeries_Rival_650_Wireless_000000000000-if01-event-mouse -> ../event8
lrwxrwxrwx. 1 root root  9 May 22 10:30 usb-SteelSeries_SteelSeries_Rival_650_Wireless_000000000000-if01-mouse -> ../mouse1
lrwxrwxrwx. 1 root root  9 May 22 10:30 usb-SteelSeries_SteelSeries_Rival_650_Wireless_000000000000-if02-event-kbd -> ../event9
```

USB devices:
```
[#]> lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
Bus 001 Device 003: ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
Bus 001 Device 005: ID 0764:0601 Cyber Power System, Inc. PR1500LCDRT2U UPS
Bus 001 Device 006: ID 1038:1610 SteelSeries ApS SteelSeries Apex Pro
Bus 001 Device 007: ID 046d:085c Logitech, Inc. C922 Pro Stream Webcam
Bus 001 Device 008: ID 1038:1507 SteelSeries ApS SteelSeries QCK Prism
Bus 001 Device 009: ID 1038:172b SteelSeries ApS SteelSeries Rival 650 Wireless
Bus 001 Device 011: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 001 Device 012: ID 0b05:1887 ASUSTek Computer, Inc. ROG_RYUO
Bus 001 Device 013: ID 0b05:18bb ASUSTek Computer, Inc. AURA LED Controller
Bus 001 Device 014: ID 0b05:18f3 ASUSTek Computer, Inc. AURA LED Controller
Bus 001 Device 035: ID 045e:02fe Microsoft Corp. Xbox Wireless Adapter for Windows
Bus 001 Device 036: ID 045e:02ea Microsoft Corp. Xbox One Controller
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
Bus 002 Device 003: ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

Verbose info of Xbox One controller:
```
[#]> lsusb -s 001:036 -v

Bus 001 Device 036: ID 045e:02ea Microsoft Corp. Xbox One Controller
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        71 [unknown]
[... 149 lines cut ...]
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

There's literally nothing on dmesg apart from the connection information.

Steam logging output also shows nothing that looks related (can provide if needed).

I also run an strace on steam with ` -e trace=file,ioctl,poll,select,futex,socket` looking for blocking behavior (can provide if needed). I could not find anything but perhaps there could be something I'm overlooking as I've never worked with input devices on the low level before and I'm searching kind of blind.

I am not sure if this is a xone problem, a kernel problem, a steam client problem or a combination of. I had these controllers for a while but had never used them on Linux. Once I migrated my Unreal Engine game development workspace from Windows to my Linux workstation and got into testing my project's controller support I hit into this from the get go. I do not remember the exact kernel version I first stumbled upon this but it was certainly a 6.13.x

Lawstorant, 2025-08-03:
Try as I might, I cannot reproduce your issue and from the description, this really sound like something is wrong with steam input in desktop mode. I'd disable steam input completely and re-test.

Does it still happen?

Nodens-, 2025-08-09:
@Lawstorant after testing, you are correct. As long as the option Steam Input for Xbox controllers is off, this does not happen. I will open an issue on steam and reference this, see maybe if Valve can pinpoint where exactly the problem is as Steam Input is required for controller to work properly with some games.

closed 2025-08-09 by Lawstorant