dlundqvist/xone #58: Audio crackle from output devices connected by Wireless Adapter

gabefraser, 2025-06-02:
Been getting audio crackle since the beginning of 2025. I've had to revert back to version [outside of this project](https://github.com/gabefraser/xonedo) which has since been abandoned.

Updating pipewire buffers/quantum sizes doesn't affect anything.

Happy to provide the correct logs, just not sure where fetch them from.

ykpmusicstudio, 2025-06-07:
> Been getting audio crackle since the beginning of 2025. I've had to revert back to version [outside of this project](https://github.com/gabefraser/xonedo) which has since been abandoned to prevent the crackle. I think version 0.3 which the fork is based on.

The 1st commit of 2025 in gamepad.c was https://github.com/dlundqvist/xone/commit/5628043b009b8cfce15fc959de03ae13c834e1a1 which is about handling out of order messages. You could try to revert to figure out if it is the origin of your issue.

> 
> This occurs on:
> 
>     * Xbox Wireless Headset
> 
>     * Audio jack from Series S/X controllers. (Tested multiple different controllers, all had the same experience).
> 
> 
> Updating pipewire buffers/quantum sizes doesn't affect anything.
> 
> Happy to provide the correct logs, just not sure where fetch them from.

I generally use `sudo dmesg -t --time-format delta`. `journalctl --system -g xone` should work as well.

Lawstorant, 2025-12-06:
Okay I went through all the commits made on top of medusalix repository and there's nothing that changes actual audio samples handling. @gabefraser is this something that still occurs?