dlundqvist/xone #59: Cannot connect dongle with Razer Thresher for Xbox One

CaioEmPessoa, 2025-06-02:
I have a Razer Thresher for Xbox One and it isn't pairing to my dongle. I think the devices are talking to each other, but it can't finish the connection for some reason.
When I try to pair them both the dongle and headset start blinking, but suddenly they start blinking slower, until they give up.

Here's the log:
```
xone-gip gip0: gip_process_pkt_chunked: offset=0x0080, length=0x0000
xone-gip gip0: gip_parse_firmware_versions: version=1.0
xone-gip gip0: gip_parse_firmware_versions: version=4.159
xone-gip gip0: gip_parse_audio_formats: formats=09-10
xone-gip gip0: gip_parse_capabilities: out=01-02-03-04-06-08-60
xone-gip gip0: gip_parse_capabilities: in=01-04-05-06-08-60
xone-gip gip0: gip_parse_classes: class=Windows.Xbox.Input.Headset
xone-gip gip0: gip_parse_interfaces: guid=a3d125bc-4ec2-9249-9dda-ef4f123ef5dc
xone-gip gip0: gip_parse_interfaces: guid=56ff7697-fd9b-8145-ad45-b645bba526d6
xone-gip-headset gip0.0: gip_register_client: registered
xone-gip-headset gip0.0: gip_headset_config: format=0x09/0x10
xone-gip gip0: gip_make_audio_config: rate=24000/1, buffer=384
xone-gip gip0: gip_make_audio_config: rate=48000/2, buffer=1536
xone-gip gip0: gip_process_pkt_chunked: offset=0x0006, length=0x0000
xone-gip gip0: gip_init_chunk_buffer: command=0x06, length=0x005a
xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x0020
xone-gip gip0: gip_process_pkt_chunked: offset=0x005a, length=0x0000
xone-gip gip0: gip_init_chunk_buffer: command=0x06, length=0x0339
xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x00ae, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x00e8, length=0x0035
xone-gip gip0: gip_process_pkt_chunked: offset=0x011d, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x0157, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x0191, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x01cb, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x0205, length=0x0034
xone-gip gip0: gip_process_pkt_chunked: offset=0x0239, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x0273, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x02ad, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x02e7, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x0321, length=0x0018
xone-gip gip0: gip_process_pkt_chunked: offset=0x0339, length=0x0000
xone-gip gip0: gip_init_chunk_buffer: command=0x06, length=0x0112
xone-gip gip0: gip_handle_pkt_acknowledge: all chunks sent
xone-dongle 3-3:1.0: xone_dongle_handle_loss: wcid=1
xone-dongle 3-3:1.0: xone_dongle_remove_client: wcid=1, address=7e:ed:8f:5f:7c:6f
xone-gip gip0: gip_destroy_adapter: unregistered
xone-gip gip0: gip_create_adapter: registered
xone-dongle 3-3:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:8f:5f:7c:6f
xone-gip gip0: gip_get_client: initialized client 0
xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:8f:5f:7c:6f, vendor=0x1532, product=0x0a16
xone-gip gip0: gip_handle_pkt_announce: firmware=4.159.19.0, hardware=1.1.1.1
xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x0080
xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x000c
xone-gip gip0: gip_process_pkt_chunked: offset=0x0006, length=0x0000
xone-gip gip0: gip_process_pkt_chunked: conflicting packet
xone-dongle 3-3:1.0: xone_dongle_process_buffer: process failed: -114
xone-dongle packet: 44 00 c0 4a 92 60 02 00 01 00 20 00 31 00 07 40
xone-dongle packet: b8 c3 a5 02 40 2f 7c 9e 00 67 b8 c3 00 00 00 00
xone-dongle packet: 00 00 00 00 88 31 a0 00 62 45 bd 11 f3 a3 7e ed
xone-dongle packet: 8f 5f 7c 6f 62 45 bd 11 f3 a3 30 00 01 00 00 00
xone-dongle packet: 06 a0 20 00 06 00 00 00 44 00 c0 4a
xone-dongle 3-3:1.0: xone_dongle_complete_in: process failed: -114

```
then it loops again

Additional info:
OS: Debian GNU/Linux 12 (bookworm) x86_64
Kernel: 6.1.0-37-amd64
Xbox One Wireless Adapter Model 1790
Razer Thresher for Xbox One

ykpmusicstudio, 2025-06-07:
> how could I know if the packet is corrupted? I tried reinstalling it all some times and it did not resolve. What kind of transmission could be causing this?

Actually my question was badly asked : is it a systematic behavior or does it works at times ?
Could you post the log with the timings as well ? It may help understanding when the packet offset=6/size=0 arrives with regards to the whole chunked transmission.

ykpmusicstudio, 2025-06-14:
I may have and idea there. It looks alike the issue I had with a gamepad with the way xone manages the fragmented (chunks) usb packets, which breaks authentication process with some devices.
Here it looks like authentication times out after a second or so, which is a behavior that makes me think that the device does not like the way the long auth packet is sent.
Then the unknown behavior happens. It can be related to the previous failure and the device was stuck into acknowledging auth packet and sends a spurious ack for command 0x6.

I may be wrong, but if not you can try the following PR (https://github.com/dlundqvist/xone/pull/46) an report if it helped or not. I suspect many devices handles the fragmented packets very strictly whereas xone is a bit permissive.

Note that the PR is still under review at this point, but if it also helps here I will try to make it go forward.

CaioEmPessoa, 2025-06-16:
I uninstalled the current xone package, and installed the one on the PR #46 . Actually, before you recommended that I had already tried this, because the problem was looking kind of the same, I got some different errors but ignored it, since the PR was still open. I did the process again now, and still got the errors, here is the log: 

(its a lot of stuff, i don't know If I printed everything since it keeps looping, if you feel that something is missing just ask me!)
```
23:47:03: xone-dongle 3-3:1.0: xone_dongle_toggle_pairing: enabled=0
23:47:04: xone-gip gip0: gip_create_adapter: registered
23:47:04: xone-dongle 3-3:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:8f:5f:7c:6f
23:47:04: xone-gip gip0: gip_get_client: initialized client 0
23:47:04: xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:8f:5f:7c:6f, vendor=0x1532, product=0x0a16
23:47:04: xone-gip gip0: gip_handle_pkt_announce: firmware=4.159.19.0, hardware=1.1.1.1
23:47:04: xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x0080
23:47:04: xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
23:47:04: xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
23:47:04: xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x000c
23:47:04: xone-gip gip0: gip_process_pkt_chunked: last chunked packet
23:47:09: xone-dongle 3-3:1.0: xone_dongle_remove_client: wcid=1, address=7e:ed:8f:5f:7c:6f
23:47:09: xone-gip gip0: gip_destroy_adapter: unregistered
23:47:09: xone-gip gip0: gip_create_adapter: registered
23:47:09: xone-dongle 3-3:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:8f:5f:7c:6f
23:47:09: xone-gip gip0: gip_get_client: initialized client 0
23:47:09: xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:8f:5f:7c:6f, vendor=0x1532, product=0x0a16
23:47:09: xone-gip gip0: gip_handle_pkt_announce: firmware=4.159.19.0, hardware=1.1.1.1
23:47:10: xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x0080
23:47:10: xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
23:47:10: xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
23:47:10: xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x000c
23:47:10: xone-gip gip0: gip_process_pkt_chunked: last chunked packet
23:47:10: xone-gip gip0: gip_process_pkt_chunked: offset=0x0080, length=0x0000
23:47:10: xone-gip gip0: gip_parse_firmware_versions: version=1.0
23:47:10: xone-gip gip0: gip_parse_firmware_versions: version=4.159
23:47:10: xone-gip gip0: gip_parse_audio_formats: formats=09-10
23:47:10: xone-gip gip0: gip_parse_capabilities: out=01-02-03-04-06-08-60
23:47:10: xone-gip gip0: gip_parse_capabilities: in=01-04-05-06-08-60
23:47:10: xone-gip gip0: gip_parse_classes: class=Windows.Xbox.Input.Headset
23:47:10: xone-gip gip0: gip_parse_interfaces: guid=a3d125bc-4ec2-9249-9dda-ef4f123ef5dc
23:47:10: xone-gip gip0: gip_parse_interfaces: guid=56ff7697-fd9b-8145-ad45-b645bba526d6
23:47:10: xone-gip gip0.0: gip_register_client: registered
23:47:10: xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:8f:5f:7c:6f, vendor=0x1532, product=0x0a16
23:47:10: xone-gip gip0: gip_handle_pkt_announce: firmware=4.159.19.0, hardware=1.1.1.1
23:47:10: xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x0080
23:47:10: xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
23:47:10: xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
23:47:10: xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x000c
23:47:10: xone-gip gip0: gip_process_pkt_chunked: last chunked packet
23:47:10: xone-gip gip0: gip_process_pkt_chunked: offset=0x0080, length=0x0000
23:47:10: xone-gip gip0: gip_handle_pkt_identify: already identified
23:47:10: xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
23:47:10: xone-gip gip0: gip_process_pkt_chunked: buffer not allocated
23:47:10: xone-dongle 3-3:1.0: xone_dongle_process_buffer: process failed: -71
23:47:10: xone-dongle packet: 7c 00 08 00 92 60 00 00 01 00 5a 00 a4 00 06 40
23:47:10: xone-dongle packet: d1 ca ac 02 40 3e c1 a9 00 63 d1 ca 00 00 00 00
23:47:10: xone-dongle packet: 00 00 00 00 88 31 3c 00 62 45 bd 11 f3 a3 7e ed
23:47:10: xone-dongle packet: 8f 5f 7c 6f 60 45 bd 11 f3 a3 a0 00 04 00 00 00
23:47:10: xone-dongle packet: 04 a0 06 ba 00 3a 06 01 04 05 06 08 60 01 1a 00
23:47:10: xone-dongle packet: 57 69 6e 64 6f 77 73 2e 58 62 6f 78 2e 49 6e 70
23:47:10: xone-dongle packet: 75 74 2e 48 65 61 64 73 65 74 02 a3 d1 25 bc 4e
23:47:10: xone-dongle packet: c2 92 49 9d da ef 4f 12 3e f5 dc 56 ff 76 97 fd
23:47:10: xone-dongle packet: 7c 00 08 00
23:47:10: xone-dongle 3-3:1.0: xone_dongle_complete_in: process failed: -71
23:47:10: xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x000c
23:47:10: xone-gip gip0: gip_process_pkt_chunked: buffer not allocated
23:47:10: xone-dongle 3-3:1.0: xone_dongle_process_buffer: process failed: -71
23:47:10: xone-dongle packet: 50 00 08 00 92 60 00 00 01 00 2c 00 b4 00 06 40
23:47:10: xone-dongle packet: d1 ca ac 02 40 42 b6 a9 00 63 d1 ca 00 00 00 00
23:47:10: xone-dongle packet: 00 00 00 00 88 11 3c 00 62 45 bd 11 f3 a3 7e ed
23:47:10: xone-dongle packet: 8f 5f 7c 6f 60 45 bd 11 f3 a3 b0 00 04 00 00 00
23:47:10: xone-dongle packet: 04 b0 06 8c 00 74 9b 81 45 ad 45 b6 45 bb a5 26
23:47:10: xone-dongle packet: d6 00 00 00 50 00 08 00
23:47:10: xone-dongle 3-3:1.0: xone_dongle_complete_in: process failed: -71
[... cut ...]
23:47:39: xone-gip gip0: gip_init_chunk_buffer: already initialized (in)
23:47:39: xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x0080
23:47:39: xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
23:47:39: xone-gip gip0: gip_process_pkt_chunked: offset=0x003a, length=0x003a
23:47:39: xone-gip gip0: gip_process_pkt_chunked: offset=0x0074, length=0x000c
23:47:39: xone-gip gip0: gip_process_pkt_chunked: last chunked packet
23:47:39: xone-gip gip0: gip_process_pkt_chunked: offset=0x0080, length=0x0000

```

ykpmusicstudio, 2025-06-16:
I could have been interesting if you managed to get the very initialization phase of the driver. At the point of you logs, the host & device are stuck in a IDENTIFY / AUTHENTICATE loop as you noticed.

This behavior is very similar to https://github.com/dlundqvist/xone/issues/42 which may be related to the order in which initialization is done in the current implementation of headset.c.

**Note:**
On the other hand, the behavior of your dongle is unique with those "conflicting packet" events (which are beyond my knowledge to analyze, since I know very little about the mt72 firmware), and at some point the device sent `gip_handle_pkt_audio_format: rejected: 0x00/0x00` with the device refusing the audio format it had previously accepted. Which is very confusing, and reminiscent of a bad RF link. It certainly does not help, but It may well be irrelevant to the main issue, though.

CaioEmPessoa, 2025-06-17:
so.. I'm sorry but how could I help? It would be needed the logs of the initialization of the driver itself? How could I get that? I'm getting my logs with journalctl -fk, so I would need to reinstall the drivers with that on another console?  

And another thing is, the problem can be specific from my dongle? If I tested with a xbox controller or something it would help?

ykpmusicstudio, 2025-06-18:
> so.. I'm sorry but how could I help? It would be needed the logs of the initialization of the driver itself? How could I get that? I'm getting my logs with journalctl -fk, so I would need to [...]

I made a small change in https://github.com/dlundqvist/xone/pull/46 to strictly comply with the spec (send Set Device : Stop before IDENTIFY). You can try the updated PR, and capture the very 1st exchange between the driver and the device if you can. Afterwards, when it's stuck in the loop, there is very useful information.

I do not expect my commit to have fixed you issue, but it may help.

ykpmusicstudio, 2025-06-22:
Thanks for this log. I see that the driver and the device are having issue as soon as the 1st INDENTIFY message is exchanged. Actually it is interrupted before the driver had a chance to decode the contents of the message => the headset disconnects during the ACK of the last chunk. Then the logs shows various behaviors of the device either keeping sending ANNOUNCE packets when in the middle of the IDENTIFY phase or resending IDENTIFY packets when the driver is already processing it.

The probable cause may be that your device is taking full use of the reliable mesage ACK workflow whereas xone driver has a very simplified automata of this workflow. This may explain why the devices either do spurious retransmissions or plain disconnections during reliable data transfers (aka chunked in xone driver terminology).

This behavior is different from other headset issues I have discussed. Here is my plan: I will try to finalize https://github.com/dlundqvist/xone/pull/46 as soon as possible then open an new PR to investigate your specific issue and probably https://github.com/dlundqvist/xone/issues/42 as well.

I will get in touch with you and @Audience9986 when I'm ready to try new things. I predict that we will go through some pain and sweat to get your headset work as it may need deep changes in the actual GIP protocol implementation.

ykpmusicstudio, 2025-06-22:
I made small changes to https://github.com/dlundqvist/xone/pull/46 in order to get more information from the logs. Can you try again and send me the logs of the initial phase ?

`dmesg --reltime` or `journalctl -k -g xone -o short-precise` would be ideal to get some more precision about message timing since there are various timeout values to track in the GIP protocol.

CaioEmPessoa, 2025-06-22:
If you want to you can messege me on discord so we can respond ourselves faster. My username is the same as on github (caioempessoa). And again, thank you for all this trouble!

Here are the logs with the updated code:
```
17:40:50.894207: xone_gip: loading out-of-tree module taints kernel.
17:40:50.894227: xone_gip: module verification failed: signature and/or required key missing - tainting kernel
17:40:52.345526: xone-dongle 3-3:1.0: xone_dongle_init: trying to load firmware xow_dongle_045e_02fe.bin
17:40:52.345716: xone-dongle 3-3:1.0: firmware: failed to load xow_dongle_045e_02fe.bin (-2)
17:40:52.345857: xone-dongle 3-3:1.0: firmware: failed to load xow_dongle_045e_02fe.bin (-2)
17:40:52.345977: xone-dongle 3-3:1.0: Direct firmware load for xow_dongle_045e_02fe.bin failed with error -2
17:40:52.346093: xone-dongle 3-3:1.0: xone_dongle_init: trying to load firmware xow_dongle.bin
17:40:52.346214: xone-dongle 3-3:1.0: firmware: direct-loading firmware xow_dongle.bin
17:40:52.346335: xone-dongle 3-3:1.0: xone_mt76_load_firmware: resetting firmware...
17:40:52.346453: xone-dongle 3-3:1.0: xone_mt76_init_radio: id=0x7613
17:40:52.346622: xone-dongle 3-3:1.0: xone_mt76_init_address: fuse_address=6c:5d:3a:11:f3:a3
17:40:52.346745: xone-dongle 3-3:1.0: xone_mt76_init_address: address=62:45:bd:11:f3:a3
17:40:52.346869: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=1, power=31
17:40:52.346988: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=6, power=32
17:40:52.347107: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=11, power=32
17:40:52.347221: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=36, power=44
17:40:52.347340: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=40, power=44
17:40:52.347458: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=44, power=43
17:40:52.347575: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=48, power=43
17:40:52.347692: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=149, power=42
17:40:52.347805: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=153, power=42
17:40:52.347921: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=157, power=42
17:40:52.348040: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=161, power=42
17:40:52.348156: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=165, power=42
17:40:52.348271: xone-dongle 3-3:1.0: xone_mt76_init_channels: channel=36
17:40:52.348973: usbcore: registered new interface driver xone-dongle
17:43:00.739316: xone-dongle 3-3:1.0: xone_mt76_suspend_radio: suspended
19:18:08.511301: xone-dongle 3-3:1.0: xone_mt76_resume_radio: resumed
19:18:08.511603: xone-dongle 3-3:1.0: xone_dongle_toggle_pairing: enabled=1
19:18:38.979302: xone-dongle 3-3:1.0: xone_dongle_toggle_pairing: enabled=0
19:19:04.303310: usbcore: deregistering interface driver xone-dongle
19:26:40.575890: xone-dongle 3-3:1.0: xone_dongle_init: trying to load firmware xow_dongle.bin
19:26:40.576205: xone-dongle 3-3:1.0: firmware: direct-loading firmware xow_dongle.bin
19:26:40.576412: xone-dongle 3-3:1.0: xone_mt76_send_firmware: build=201703281033____
19:26:40.576569: xone-dongle 3-3:1.0: xone_mt76_init_radio: id=0x7613
19:26:40.576716: xone-dongle 3-3:1.0: xone_mt76_init_address: fuse_address=6c:5d:3a:11:f3:a3
19:26:40.576853: xone-dongle 3-3:1.0: xone_mt76_init_address: address=62:45:bd:11:f3:a3
19:26:40.577014: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=1, power=31
19:26:40.577167: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=6, power=32
19:26:40.577321: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=11, power=32
19:26:40.577482: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=36, power=44
19:26:40.577646: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=40, power=44
19:26:40.577810: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=44, power=43
19:26:40.577972: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=48, power=43
19:26:40.578138: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=149, power=42
19:26:40.578302: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=153, power=42
19:26:40.578436: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=157, power=42
19:26:40.578557: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=161, power=42
19:26:40.578675: xone-dongle 3-3:1.0: xone_mt76_evaluate_channels: channel=165, power=42
19:26:40.578818: xone-dongle 3-3:1.0: xone_mt76_init_channels: channel=36
19:26:40.578940: usbcore: registered new interface driver xone-dongle
19:27:00.035302: xone-dongle 3-3:1.0: xone_dongle_toggle_pairing: enabled=1
19:27:22.155295: xone-dongle 3-3:1.0: xone_dongle_pair_client: address=7e:ed:8f:5f:7c:6f
19:27:22.156392: xone-dongle 3-3:1.0: xone_dongle_toggle_pairing: enabled=0
19:27:23.899313: xone-gip gip0: gip_create_adapter: registered
19:27:23.899565: xone-dongle 3-3:1.0: xone_dongle_add_client: wcid=1, address=7e:ed:8f:5f:7c:6f
19:27:23.907296: xone-gip gip0: gip_get_client: initialized client 0
19:27:23.907472: xone-gip gip0: gip_handle_pkt_announce: address=7e:ed:8f:5f:7c:6f, vendor=0x1532, product=0x0a16
19:27:23.907617: xone-gip gip0: gip_handle_pkt_announce: firmware=4.159.19.0, hardware=1.1.1.1
19:27:23.923307: xone-gip gip0: gip_init_chunk_buffer: command=0x04, length=0x0080
19:27:23.923574: xone-gip gip0: gip_process_pkt_chunked: offset=0x0000, length=0x003a
[... cut ...]
19:27:45.793031: xone-gip gip0: gip_parse_interfaces: guid=a3d125bc-4ec2-9249-9dda-ef4f123ef5dc
19:27:45.793137: xone-gip gip0: gip_parse_interfaces: guid=56ff7697-fd9b-8145-ad45-b645bba526d6
19:27:45.793254: xone-gip-headset gip0.0: gip_register_client: registered
19:27:46.815305: xone-gip-headset gip0.0: gip_headset_config: format=0x09/0x10
19:27:46.827297: xone-gip gip0: gip_make_audio_config: rate=24000/1, buffer=384
19:27:46.827469: xone-gip gip0: gip_make_audio_config: rate=48000/2, buffer=1536

```

closed 2025-06-27 by Lawstorant