dlundqvist/xone #61: xone-dongle not loading after restart

tarteens, 2025-06-09:
Hi, 

i've insalled this version of xone driver. After a bit of a struggle a made the pairing and my XBox One controller working (using the official dongle). 

However after a reboot it's not getting recognized, i have to unplug/replug the dongle. 

In the dmesg I have those errors : 

`
[    2.776574] xone-dongle 3-2:1.0: Direct firmware load for xow_dongle_045e_02fe.bin failed with error -2
[    2.776613] xone-dongle 3-2:1.0: Direct firmware load for xow_dongle.bin failed with error -2
[    2.776615] xone-dongle 3-2:1.0: xone_dongle_init: request firmware failed: -2
[    2.777651] xone-dongle 3-2:1.0: probe with driver xone-dongle failed with error -2
[    2.777684] usbcore: registered new interface driver xone-dongle
`

I'm not sure to know what's need to be done. 
Thanks in advance for your help.

EndaevorOs , with kernel 6.14.10-arch1-1 , followed the manual installation.

ykpmusicstudio, 2025-06-09:
> In the dmesg I have those errors :
> 
> [ 2.776574] xone-dongle 3-2:1.0: Direct firmware load for xow_dongle_045e_02fe.bin failed with error -2
> 
> I'm not sure to know what's need to be done. Thanks in advance for your help.
> 
> EndaevorOs , with kernel 6.14.10-arch1-1 , followed the manual installation.

This error means that the dongle firmware was not properly downloaded => See https://github.com/dlundqvist/xone/issues/62 as it may well will fix your issue.

If this works, you can thank https://github.com/kstrehan since he just figured it out.

Sajito, 2025-06-10:
More or less same issue here. My log contains only this line
```
[ 1762.477259] xone-dongle 7-1.4.4.1.2:1.0: Direct firmware load for xow_dongle_045e_02fe.bin failed with error -2
```

Firmware binary is present.
```
$ ls -la /lib/firmware/xow*
.rw-r--r-- 71k root 10 Jun 15:50  /lib/firmware/xow_dongle.bin
.rw-r--r-- 70k root 10 Jun 15:50  /lib/firmware/xow_dongle_045e_02e6.bin
```

Edit: Nevermind. After a removing and reinstalling the driver, the error mentioned is still logged, yet the controller works again.

tarteens, 2025-06-10:
> This error means that the dongle firmware was not properly downloaded => See [#62](https://github.com/dlundqvist/xone/issues/62) as it may well will fix your issue.
>
> If this works, you can thank [...]

Unfortunatly i still have to unplug / replug the dongle...
I guess it will have to suffice for now.

MisterGoodcat, 2025-06-21:
I'll try that today. 

Please note that sleep (suspend) is not a problem at all, the controller keeps working even the next day. It's when the computer is shut down and rebooted that the controller needs a physical unplug every time.

Lawstorant, 2025-06-21:
Got it, i just thought that maybe not trying to load inexistent firmware may help.

I'll test out shutdowns then

MisterGoodcat, 2025-06-21:
That was a bit of a journey. Never had set up DKMS, never had enrolled a MOK key for it (I have secure boot enabled), and then at first reboot after installing the new version, the dongle did not work at all...

Long story short: I think the issue has been resolved. After another reboot, everything worked as expected. I did quite some testing, for example booting with the controller powered on and searching for the dongle already, with the controller powered off and turning it on later etc. - it worked every time. Thank you so much!

I'm fine with leaving it as-is and use the DKMS setup, but would you expect any issues with akmod? Most Fedora fellows would probably come from there (copr/lpf).

Lawstorant, 2025-06-21:
Is fedora building xone? where can I find the package?

MisterGoodcat, 2025-06-21:
Interesting, and confusing actually.

I originally found xone on Fedora Copr ("COmmunity PRojects"), which is just a build platform for unofficial packages. I assumed that this was related to your project here because it's being referred to in several places, but now realize that this is not the case, apologies for that.

Upon closer inspection, my understanding is that the project at Copr provides the meta information and spec files for how to turn the xone sources into actual RPM packages for Fedora. It adds additional features on top of that, for example bundling the xpad-noone fork. The output is a kmod (not DKMS), and that was what I've been using until now. 

Some links:

The repo where users find the Fedora-specific install/uninstall instructions: https://copr.fedorainfracloud.org/coprs/sentry/xone/
The repo with the actual sources for the above: https://pagure.io/xone-rpm
Maintainer (I assume): https://pagure.io/user/sentry

If you look at the sources you'll find that they apparently made the switch from the upstream medusalix project to your fork only recently.

Hope this helps!

Lawstorant, 2025-06-21:
Yes! This helps a lot since I don't really know how it works in Fedora word. At least it seems that COPR is basically like AUR for Arch-based distros so just community.

And yeah, the original repo has been inactive for more than a year and most packagers switched to this one. We might as well call this one upstream. Just like real rivers, their streams are not fixed to one place :P

MisterGoodcat, 2025-06-21:
You can actually ask GitHub support to detach your fork and turn it into a standalone repo to keep all its current status (instead of manually cloning it and deleting the fork, which is like starting anew). I personally think it's a legitimate move to do that and just give credit for prior work, if upstream was abandoned or the fork has just deviated so much that clearly the changes will never be merged back.

MisterGoodcat, 2025-06-30:
Here's a quick follow-up: after the update to kernel 6.15.4 the controller stopped working. The module was activated for the kernel automatically, but in dmesg the following errors were logged:

```
[   22.072717] xone-dongle 1-3:1.0: xone_dongle_init: init radio failed: -110
[   22.079278] xone-dongle 1-3:1.0: probe with driver xone-dongle failed with error -110
[   22.079404] usbcore: registered new interface driver xone-dongle
```

Running uninstall then install again fixed it.

MisterGoodcat, 2025-07-18:
Might have been some hiccup last time, the last kernel update was smooth, module was auto-reinstalled and the controller kept working without any manual intervention.

Lawstorant, 2025-07-18:
Nah, there indeed is some underlying issue. I managed to catch a 110 error after boot and I had to replug the dongle.

Lawstorant, 2025-07-31:
Hmm, I'll try doing some work around maybe trying to load the firmware later. We'll see if it will help.

Lawstorant, 2025-07-31:
This IS the place to post it. Thanks! This is further proof that something about early firmware loading is just not working.

autisticgandalf, 2025-08-02:
I've tried the fix you've made and it didn't work for me

just in case this is exactly what i've done,
first i've uninstall xone 3.5 and reboot
then i've redownload 3.5 and apply the change you've made by replace the changed files with the ones you've made
i've installed this version with no issue, no error.
i've downloaded th firmware, no issue, no error
i've plugged the dongle and done a `sudo dmesg | grep xone`
here is the results :
```
[    7.733805] usbcore: registered new interface driver xone-dongle
[  146.842780] xone-dongle 3-2:1.0: xone_dongle_init: trying to load firmware xow_dongle.bin
[  146.852872] xone-dongle 3-2:1.0: xone_mt76_send_firmware: build=201703281033____
[  146.985138] xone-dongle 3-2:1.0: xone_mt76_init_radio: id=0x7613
[  147.014370] xone-dongle 3-2:1.0: xone_mt76_init_address: fuse_address=6c:5d:3a:12:71:b9
[  147.014374] xone-dongle 3-2:1.0: xone_mt76_init_address: address=62:45:bd:12:71:b9
[  147.018910] xone-dongle 3-2:1.0: xone_mt76_evaluate_channels: channel=1, power=30
[  147.020785] xone-dongle 3-2:1.0: xone_mt76_evaluate_channels: channel=6, power=32
[  147.022660] xone-dongle 3-2:1.0: xone_mt76_evaluate_channels: channel=11, power=32
[  147.024535] xone-dongle 3-2:1.0: xone_mt76_evaluate_channels: channel=36, power=44
[  147.026410] xone-dongle 3-2:1.0: xone_mt76_evaluate_channels: channel=40, power=44
[  147.028285] xone-dongle 3-2:1.0: xone_mt76_evaluate_channels: channel=44, power=43
[  147.030159] xone-dongle 3-2:1.0: xone_mt76_evaluate_channels: channel=48, power=43
[  147.032035] xone-dongle 3-2:1.0: xone_mt76_evaluate_channels: channel=149, power=40
[  147.033911] xone-dongle 3-2:1.0: xone_mt76_evaluate_channels: channel=153, power=40
[  147.035785] xone-dongle 3-2:1.0: xone_mt76_evaluate_channels: channel=157, power=40
[  147.037660] xone-dongle 3-2:1.0: xone_mt76_evaluate_channels: channel=161, power=40
[  147.039535] xone-dongle 3-2:1.0: xone_mt76_evaluate_channels: channel=165, power=40
[  147.039870] xone-dongle 3-2:1.0: xone_mt76_init_channels: channel=36

```
no error, my Xbox series controller has no problem to work at all, i don't see any issues
then i reboot and done a `sudo dmesg | grep xone`
```
[    6.549881] xone-dongle 3-2:1.0: xone_dongle_init: trying to load firmware xow_dongle.bin
[    6.549915] xone-dongle 3-2:1.0: Direct firmware load for xow_dongle.bin failed with error -2
[    6.549917] xone-dongle 3-2:1.0: xone_dongle_init: request firmware failed: -2
[    6.550945] xone-dongle 3-2:1.0: probe with driver xone-dongle failed with error -2
[    6.550967] usbcore: registered new interface driver xone-dongle

```
and it don't work, i re-plug the dongle and it work, so same issue than before
if you need more information don't hesitate

Lawstorant, 2025-08-02:
Yeah, I don't think you actually applied the changes. The debug messages don't match with the ones I changed/added in the fix branch.

I'll push the branch directly to this repo instead of my fork so you'll be able to switch easily.

autisticgandalf, 2025-08-02:
so i've tried with `xone-dkms-git` and `xone-dongle-firmware` from AUR
and same issue

```
[    7.095375] xone-dongle 3-2:1.0: Direct firmware load for xow_dongle.bin failed with error -2
[    8.119284] xone-dongle 3-2:1.0: Direct firmware load for xow_dongle.bin failed with error -2
[    9.143581] xone-dongle 3-2:1.0: Direct firmware load for xow_dongle.bin failed with error -2
[   10.167530] xone-dongle 3-2:1.0: xone_dongle_init: request firmware failed: -2
[   10.168638] xone-dongle 3-2:1.0: probe with driver xone-dongle failed with error -2
[   10.168712] usbcore: registered new interface driver xone-dongle

```

Lawstorant, 2025-08-02:
Are you running running an encrypted system by chance? LVM on crypt or maybe crypt on LVM?