dlundqvist/xone #73: Kernel Panic with 6.16-rc3

wantija, 2025-06-25:
Title is self explanitory, getting a kernel panic with  that kernel, 6.15 is fine. The error that gets displayed on the kernel panic is: `stack_protector: Kernel stack is corrupted in gip_auth_get_transcript+0x01/0x90 [xone_gip]`. Happens immediatley when wirelessly connecting an Xbox One controller and also via wired.

Lawstorant, 2025-06-25:
I'm unsure what is this caused by, but seems like 6.16-rc3 is broken internally. Apart from xone, iwlwifi is completely broken and some other modules spam with errors as well.

Lawstorant, 2025-06-30:
Still happens with RC4. Indeed the problem lies within `gip_auth_get_transcript()`. I removed all of it's contents and just `return 0` and no more panics.

Lawstorant, 2025-06-30:
Yeah, upstream changes without proper documentation. `crypto_shash_export` now returns more data and we're getting buffer overflows.

I think for our sanity, we should switch to `lib/crypto` soon

PR #89 (2025-07-29): Fix kernel panic on Linux 6.16
Fixes #88
Fixes #73
closed 2025-07-29 by Lawstorant