dlundqvist/xone #83: Impulse trigger/Trigger rumble not working

Rishikant181, 2025-07-05:
I own a GameSir G7 SE (Wired) and using this driver on Fedora 42, Kernel 6.15.4-200.fc42.x86_64.

My issue is impulse trigger/trigger rumble does not work in games, even though both the hardware and the games supports it.

I don't know if this is supposed to work on Linux at all or if it requires any additional configuration.

Thanks!

Rishikant181, 2025-07-05:
[This](https://www.pcgamingwiki.com/wiki/List_of_games_that_support_Xbox_One_impulse_triggers) provides a list of games that support impulse triggers on PC

Lawstorant, 2025-07-05:
I don't think this list is right at all. I was playing Split/Fiction on windows and there were no impulse triggers there.

The game list is supper small + most of it is very small stuff. If someone ever decides to try and support this, then maybe it'll happen but this won't be planned for the foreseeable future.

Lawstorant, 2025-11-08:
Okay guys, this won't be supported at the moment. The driver itself can send the comamnds for trigger rumble but current Linux FF api just doesn't support this. I may try to extend the linux FF api for that, but it won't come for months so for now, I'm closing.

closed 2025-11-08 by Lawstorant

motolav, 2026-04-03:
A few years ago, I had found on the LKML that someone did implement trigger rumble for the xbox one controllers but it seems like with how it was sent in the patches got ignored, I had shared what I found on the xpad github.

[[PATCH 0/4] input: Implement an API for trigger rumble motors](https://lore.kernel.org/lkml/20220410220449.5071-1-dbdaniel42@gmail.com/)
[[PATCH 1/4] input: uapi: Add trigger_left and trigger_right to ff_rumble_effect struct](https://lore.kernel.org/lkml/20220410220633.5235-1-dbdaniel42@gmail.com/)
[[PATCH 2/4] input: ff-memless: Add trigger left/right in ml_combine_effects](https://lore.kernel.org/lkml/20220410220649.5258-1-dbdaniel42@gmail.com/)
[[PATCH 3/4] input: xpad: Implement trigger rumble for Xbox One controllers](https://lore.kernel.org/lkml/20220410220701.5279-1-dbdaniel42@gmail.com/)
[[PATCH 4/4] input: hid-microsoft: Implement trigger rumble for Xbox One S over bluetooth](https://lore.kernel.org/lkml/20220410220713.5303-1-dbdaniel42@gmail.com/)