dlundqvist/xone #85: Steam not able to detect controllers after suspens, reboot or power off.

isaranghi, 2025-07-11:
Hi all!
I'm running Debian 13 (Trixie) and have set up xone with the official Microsoft Xbox Wireless USB dongle. I paired two Xbox wireless controllers, and everything works perfectly in Steam Big Picture — both controllers are detected and function correctly.

However, after a reboot, shutdown, or suspend, while the controllers auto-connect to the dongle (solid light), Steam no longer detects them. Oddly, I can wake the PC from suspend using the controller's Xbox button, so the connection is clearly there. But in Steam > Controller settings, it says no controllers detected. Unplugging and replugging the dongle fixes it — Steam immediately detects both controllers again.

To compare, I uninstalled xone and installed xow. With xow, the controllers remain detected in Steam even after reboot/suspend, though the wake-from-suspend feature doesn't work.

I'd really like to use xone, as it's the newer and recommended driver, but I'm not sure why Steam doesn't detect the controllers post-boot without replugging the dongle.

Any ideas or suggestions?

Thanks in advance!

ronald-b, 2025-07-13:
I have a similar problem. My problem is that the controllers keep disconnecting, and when they connect, they aren't recognized by the system.

OS: CachyOS
Kernel: Linux 6.15.6-2-cachyos

isaranghi, 2025-07-14:
> I have a similar problem. My problem is that the controllers keep disconnecting, and when they connect, they aren't recognized by the system.
> 
> OS: CachyOS Kernel: Linux 6.15.6-2-cachyos

Yeah, I haven’t been able to find a solution either. For now, I’ve switched back to xow, since it seems to be working fine for me (aside from not being able to wake the PC from suspend — which I can live with). I’ll give xone another try in the near future.

closed 2025-07-15 by Lawstorant