dlundqvist/xone #88: 6.16 kernel panic

AlkaizerLord, 2025-07-29:
So Im on cachyOS and just updated my kernel to the new 6.16 version. I installed xone directly from the github repo using `git clone https://github.com/dlundqvist/xone
`

After cloning the repo I installed using `sudo ./install.sh`

Everything seemed fine no issues. Nothing happened at all until a few hours later once I turned on my Xbox elite controller to sync with the dongle. Thats when I got a blue screen kernel panic.

[image]

Once rebooted I had some issues with btrfs log-tree errors due what Im guessing was the kernel panic and abrupt shutdown. Easy fix just ran `btrfs recover zero-log /dev/<disk>`........Leaving this here in case anyone else is in a similar situation.

Once that was sorted out I was able to boot back up into my system. I decided to test it again and see if it was only happening as soon as I try to sync the controller to the dongle. Again immediately I got the same blue screen kernel panic.

Rebooted again and did a complete removal of xone. Rebooted just for safe measure and then checked that xone was indeed removed completely. Used `find` & `lsmod` just to make sure nothing was there and then proceeded to reinstall xone module again.

No problems everything looked good and then again immediately as I go to sync my controller to the dongle KERNEL PANIC

Lawstorant, 2025-07-29:
Yeah, we were aware in #73 but the changes in upstream linux weren't fully developed. I'll now take a al look at fixing this.

AlkaizerLord, 2025-07-29:
*derp* I dont know why I didnt look at the closed issues first before posting this. Sorry about that

closed 2025-07-29 by Lawstorant

Lawstorant, 2025-07-29:
Turns out it was an easy fix. Moreover, it's just luck that this never broke before as the sturct used as buffer was... not optimal.

Lawstorant, 2025-07-30:
@AlkaizerLord could you test the newest version to confirm?

AlkaizerLord, 2025-07-30:
Sure thing! Ill be able to test it out soon, 2 - 3 hrs. Will report back ASAP.

PR #89 (2025-07-29): Fix kernel panic on Linux 6.16
Fixes #88
Fixes #73