dlundqvist/xone #97: Validate against published GIP spec

benjaminglass1, 2025-08-03:
I couldn't find any discussion on this in the repo, so I wanted to throw out that it looks like Microsoft finally published an official spec for the GIP interface at https://learn.microsoft.com/en-us/openspecs/windows_protocols/ms-gipusb/e7c90904-5e21-426e-b9ad-d82adeee0dbc late last year. Has anyone validated the xone implementation against the spec, or is such an effort ongoing? I've been looking for an opportunity to get involved with this project (which I use regularly), and am happy to help!

Lawstorant, 2025-08-04:
I don't think so, no. The original author implemented basically the whole thing with reverse engineering and vanished more than a year ago. Currently, we're just making sure this driver is compatible with new kernels and doing fixes here and there.

@endrift seems to be working on a completely new implementation of GIP based on this document that will hopefully supersede `xone` and work better.

benjaminglass1, 2025-08-05:
@Lawstorant @endrift is this implementation somewhere I can contribute to? I don't see it on GitHub.

benjaminglass1, 2025-08-05:
@endrift I'll close this, but I would greatly appreciate a link to the patch when submitted!

endrift, 2025-08-05:
Will do.

closed 2025-08-05 by benjaminglass1

Lawstorant, 2025-08-06:
@endrift you won't create a github repo as a mirror to speed up development and issue creation? I think that's what xpad does and we sure do this with hid-universal-pidff

endrift, 2025-08-06:
There is no official xpad repo. There are a handful of unofficial ones, but none of them are close to official. Having a module be both *in* and *out* of tree is kind of tricky to manage, given that there are two independent "sources of truth" that people can individually send patches to and have get out of sync. I don't really want to manage that, and since it's just me working on it at the moment, there's not really any strong reason to put it up on its own repo.

That said, a userspace version is part of SDL now:  https://github.com/libsdl-org/SDL/blob/main/src/joystick/hidapi/SDL_hidapi_gip.c

endrift, 2025-08-08:
Submitted the patchset: https://lore.kernel.org/linux-input/20250808043017.1953101-1-vi@endrift.com/T/