dlundqvist/xone #106: Dongle firmware failing to load on Kernel 6.15.10 on Fedora 42?

theloneraven, 2025-08-22:
I was first alerted when my controller wasn't recognized by Steam at all when updating to the new kernel. After troubleshooting for hours, I found that the firmware for my dongle was failing to load even after rebuilding, reinstalling firmware, checking that secure boot was off, and making sure permissions were correct. Looking at an strace I found that xone_dongle.ko wasn't even requesting the firmware to be loaded. 

I suspect that the only explanation is that it's somehow broken on the new kernel version. I've tried everything.

Lawstorant, 2025-08-24:
No, that always shows up, even with SB off. Since there aren't any mire messages it simply seems like your dongle isn't reckognized as plugged in. Could you post the output of `lsusb`?

theloneraven, 2025-08-24:
Sorry, I actually did physically have it unplugged. I plugged it back in and tried again and actually got this error this time:
[    5.022512] xone-dongle 1-2:1.0: Direct firmware load for xow_dongle_045e_02e6.bin failed with error -2

this is the dongle in lsusb:
Bus 001 Device 003: ID 045e:02e6 Microsoft Corp. Xbox Wireless Adapter for Windows

Lawstorant, 2025-08-25:
Just one? There should be multiple.

lostiniceland, 2025-09-18:
I did provide Wireshark traces for previous issues (for the wired controller) so I should be able to do this again. Maybe we could update the documentation with a debug tutorial or even a little script that pulls out as much information as possible.

Another option for quicker testing would be a Terraform setup that quickly installs a VM via KVM and runs Ansible to provision xone. With usb passthrough you should be able to test locally. This should bring down the feedback loop to 1-2minutes.
That would be something which I could provide for instance (since I just had to setup something like this for a horribly outdated vpn client).

Lawstorant, 2025-09-22:
I just tried this in a VM and it works just fine. @lostiniceland @theloneraven make sure you have `cabextract` installed to properly extract firmware. I tried with tumbleweed as well and it load the firmware without issues, again, after installing cabextract before install (the script warns that the cabextract is not installed after install)

closed 2025-09-22 by Lawstorant

Lawstorant, 2025-09-27:
Man, you don't have the firmware in your system. No winder it won't load.

lostiniceland, 2025-09-28:
It is working now.
I had this setup running for over a year and all I had to do was running `uninstall` and `install --release` after kernel upgrades.
How could I know that the firmware got removed or needs updating.

Anyways, thank you very much.

Lawstorant, 2025-09-28:
> How could I know that the firmware got removed or needs updating.

> `Direct firmware load for xow_dongle_045e_02e6.bin failed with error -2`

```shell
tomek@laptok ~ $ errno 2    
ENOENT 2 No such file or directory
```