dlundqvist/xone #109: Controller not connecting to wireless adapter, no errors reported

filipdutescu, 2025-08-28:
# Description

The USB adapter and the controller are not connecting to each other. There are no errors reported from my initial investigation and everything should be "just working". Re-pairing is also not working, they are not able to find each other.

## Expected behaviour

Inserting the USB adapter and powering on the controller should lead to the latter connecting to the former, as it does on Windows.

## Actual behvaiour

Inserting the USB adapter is successful, powering on the controller works, but it stays in a disconnected state (power button blinking on the controller, awaiting the dongle).

# Technical information

**Distro:** Arch Linux
**Kernel:**  6.16.3.arch1-1
**Linux firmware:**  20250808-1
**xone-dkms:**  0.4.3-1 (source: <https://aur.archlinux.org/packages/xone-dkms>)
**xone-dongle-firmware:**   1.0.46.1-2 (source: <https://aur.archlinux.org/packages/xone-dongle-firmware>)

## Logs

- `dmseg`:
  ```shell
  [ 1576.327503] usb 1-6: new high-speed USB device number 3 using xhci_hcd
  [ 1576.542403] usb 1-6: New USB device found, idVendor=045e, idProduct=02fe, bcdDevice= 1.00
  [ 1576.542411] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
  [ 1576.542412] usb 1-6: Product: XBOX ACC
  [ 1576.542414] usb 1-6: Manufacturer: Microsoft Inc.
  [ 1576.542415] usb 1-6: SerialNumber: 047594
  [ 1576.590714] xone_gip: loading out-of-tree module taints kernel.
  [ 1576.590719] xone_gip: module verification failed: signature and/or required key missing - tainting kernel
  [ 1576.771363] usb 1-6: reset high-speed USB device number 3 using xhci_hcd
  [ 1577.014940] usbcore: registered new interface driver xone-dongle
  ```

- `lsusb`: shows the adapter via
  ```shell
  Bus 001 Device 003: ID 045e:02fe Microsoft Corp. Xbox Wireless Adapter for Windows
  ```

- `ls /usr/lib/firmware | grep xow`:
  ```shell
  /usr/lib/firmware/xow_dongle.bin
  /usr/lib/firmware/xow_dongle_045e_02e6.bin
  ```

Lawstorant, 2025-08-28:
Is the dongle blinking when you try to re-pair the controller?

Lawstorant, 2025-08-28:
Hell, IDK then. Maybe you have some lingering old dkms version of xone?

filipdutescu, 2025-08-28:
Cant be, my install (OS install) is fresh, has less than a week.

Lawstorant, 2025-08-28:
Did you reboot tour system after installing xone?

filipdutescu, 2025-08-28:
Yes, multiple times. Somehow, `sudo make test` fixed it though. I also checked everything I reported above after every reboot as well

closed 2025-08-28 by Lawstorant