dlundqvist/xone #110: Kernel panic when connecting controller

forest-cat, 2025-08-29:
## Description
I am not sure if I am in the correct repo, feel free to inform me when im wrong
When having the driver loaded via dkms and having rebooted, then as soon as i connect my controller via cable or via the dongle i get a kernel panic

## System information
Version: `xone/0.3.4.r0.gd31a04e, 6.16.3-arch1-1, x86_64: installed` (dkms status output)

## How to reproduce
1. Install the driver and make sure its loaded
2. Connect your dongle and the controller
3. Wait for it to connect, as soon as its connected it crashes

## Attached error log
```
[15430.330006] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=146.75.117.91 DST=192.168.140.14 LEN=103 TOS=0x00 PREC=0x00 TTL=55 ID=8036 DF PROTO=TCP SPT=443 DPT=50994 WINDOW=424 RES=0x00 ACK PSH URGP=0 
[15430.330096] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=146.75.117.91 DST=192.168.140.14 LEN=76 TOS=0x00 PREC=0x00 TTL=55 ID=8037 DF PROTO=TCP SPT=443 DPT=50994 WINDOW=424 RES=0x00 ACK PSH FIN URGP=0 
[15430.376660] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=146.75.117.91 DST=192.168.140.14 LEN=76 TOS=0x00 PREC=0x00 TTL=55 ID=8038 DF PROTO=TCP SPT=443 DPT=50994 WINDOW=424 RES=0x00 ACK PSH FIN URGP=0 
[15430.535829] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=146.75.121.91 DST=192.168.140.14 LEN=127 TOS=0x00 PREC=0x00 TTL=55 ID=47795 DF PROTO=TCP SPT=443 DPT=39776 WINDOW=226 RES=0x00 ACK PSH FIN URGP=0 
[15453.086973] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=35.186.224.44 DST=192.168.140.14 LEN=449 TOS=0x00 PREC=0x00 TTL=118 ID=30326 PROTO=TCP SPT=443 DPT=33150 WINDOW=1040 RES=0x00 ACK PSH URGP=0 
[15453.306663] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=35.186.224.44 DST=192.168.140.14 LEN=449 TOS=0x00 PREC=0x00 TTL=118 ID=30327 PROTO=TCP SPT=443 DPT=33150 WINDOW=1040 RES=0x00 ACK PSH URGP=0 
[15458.370684] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=146.75.121.91 DST=192.168.140.14 LEN=127 TOS=0x00 PREC=0x00 TTL=51 ID=7535 DF PROTO=TCP SPT=443 DPT=47220 WINDOW=201 RES=0x00 ACK PSH FIN URGP=0 
[15472.832065] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=34.107.243.93 DST=192.168.140.14 LEN=125 TOS=0x00 PREC=0x00 TTL=118 ID=63797 PROTO=TCP SPT=443 DPT=51452 WINDOW=1070 RES=0x00 ACK PSH URGP=0 
[15487.689582] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=146.75.121.91 DST=192.168.140.14 LEN=127 TOS=0x00 PREC=0x00 TTL=52 ID=47802 DF PROTO=TCP SPT=443 DPT=39776 WINDOW=226 RES=0x00 ACK PSH FIN URGP=0 
[15487.924655] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=146.75.117.91 DST=192.168.140.14 LEN=127 TOS=0x00 PREC=0x00 TTL=50 ID=8046 DF PROTO=TCP SPT=443 DPT=50994 WINDOW=424 RES=0x00 ACK PSH FIN URGP=0 
[15494.144004] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=34.107.243.93 DST=192.168.140.14 LEN=125 TOS=0x00 PREC=0x00 TTL=118 ID=63799 PROTO=TCP SPT=443 DPT=51452 WINDOW=1070 RES=0x00 ACK PSH URGP=0 
[15522.304737] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=34.107.243.93 DST=192.168.140.14 LEN=125 TOS=0x00 PREC=0x00 TTL=118 ID=63800 PROTO=TCP SPT=443 DPT=51452 WINDOW=1070 RES=0x00 ACK PSH URGP=0 
[15538.594688] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=35.186.224.44 DST=192.168.140.14 LEN=449 TOS=0x00 PREC=0x00 TTL=118 ID=30336 PROTO=TCP SPT=443 DPT=33150 WINDOW=1040 RES=0x00 ACK PSH URGP=0 
[15568.802550] [UFW BLOCK] IN=enp13s0 OUT= MAC=2c:56:dc:dc:36:a8:60:e3:27:80:e0:d9:08:00 SRC=35.186.224.44 DST=192.168.140.14 LEN=449 TOS=0x00 PREC=0x00 TTL=118 ID=30337 PROTO=TCP SPT=443 DPT=33150 WINDOW=1040 RES=0x00 ACK PSH URGP=0 
[16066.203794] perf: interrupt took too long (3165 > 3146), lowering kernel.perf_event_max_sample_rate to 63000
[18558.491482] [UFW BLOCK] IN=wg_alvern1 OUT= MAC= SRC=172.67.145.254 DST=192.168.188.205 LEN=91 TOS=0x00 PREC=0x00 TTL=57 ID=3396 DF PROTO=TCP SPT=443 DPT=41214 WINDOW=17 RES=0x00 ACK PSH URGP=0 
[18559.495537] [UFW BLOCK] IN=wg_alvern1 OUT= MAC= SRC=172.67.145.254 DST=192.168.188.205 LEN=91 TOS=0x00 PREC=0x00 TTL=56 ID=49845 DF PROTO=TCP SPT=443 DPT=41222 WINDOW=16 RES=0x00 ACK PSH URGP=0 
[18559.495564] [UFW BLOCK] IN=wg_alvern1 OUT= MAC= SRC=172.64.149.114 DST=192.168.188.205 LEN=91 TOS=0x00 PREC=0x00 TTL=56 ID=53233 DF PROTO=TCP SPT=443 DPT=38834 WINDOW=16 RES=0x00 ACK PSH URGP=0 
[18560.358656] [UFW BLOCK] IN=wg_alvern1 OUT= MAC= SRC=172.64.149.114 DST=192.168.188.205 LEN=91 TOS=0x00 PREC=0x00 TTL=56 ID=53237 DF PROTO=TCP SPT=443 DPT=38834 WINDOW=16 RES=0x00 ACK PSH FIN URGP=0 
[18563.675273] wireguard: wg_alvern2: Could not create IPv4 socket
[18566.357416] wireguard: wg_alvern2: Could not create IPv4 socket
[21816.124681] usb 1-2: new high-speed USB device number 6 using xhci_hcd
[21816.466760] usb 1-2: New USB device found, idVendor=046d, idProduct=0825, bcdDevice= 0.12
[21816.466777] usb 1-2: New USB device strings: Mfr=0, Product=0, SerialNumber=2
[21816.466785] usb 1-2: SerialNumber: 1314AA20
[21817.911646] usb 1-2: set resolution quirk: cval->res = 384
[21817.938267] videodev: Linux video capture interface: v2.00
[21817.966006] usb 1-2: Found UVC 1.00 device <unnamed> (046d:0825)
[21818.159204] usbcore: registered new interface driver uvcvideo
[22043.258984] usb 1-2: reset high-speed USB device number 6 using xhci_hcd
[22093.821986] usb 1-2: reset high-speed USB device number 6 using xhci_hcd
[22108.904153] usb 1-2: reset high-speed USB device number 6 using xhci_hcd
[22187.092458] usb 1-2: reset high-speed USB device number 6 using xhci_hcd
[22192.639473] usb 1-2: reset high-speed USB device number 6 using xhci_hcd
[27910.432938] usb 1-2: reset high-speed USB device number 6 using xhci_hcd
[27915.826678] usb 1-2: USB disconnect, device number 6
[35465.219622] perf: interrupt took too long (3985 > 3956), lowering kernel.perf_event_max_sample_rate to 50000
[42835.810062] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: gip_auth_get_transcript+0x81/0x90 [xone_gip]
[42835.810077] CPU: 6 UID: 1000 PID: 47204 Comm: ThreadPoolForeg Tainted: G           OE       6.16.3-arch1-1 #1 PREEMPT(full)  1f0972552899c74baa9bd5c16d3531764dd9a1bb
[42835.810084] Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[42835.810085] Hardware name: System manufacturer System Product Name/Z170-DELUXE, BIOS 3801 03/14/2018
[42835.810088] Call Trace:
[42835.810094]  <TASK>
[42835.810098]  dump_stack_lvl+0x5d/0x80
[42835.810111]  panic+0x119/0x2de
[42835.810121]  ? crypto_shash_finup+0x77/0x1e0
[42835.810130]  ? gip_auth_get_transcript+0x81/0x90 [xone_gip 31d51abdd4e12bbf79188657a9e679848a766a16]
[42835.810145]  __stack_chk_fail+0x19/0x20
[42835.810156]  gip_auth_get_transcript+0x81/0x90 [xone_gip 31d51abdd4e12bbf79188657a9e679848a766a16]
[42835.810164]  gip_auth_send_pkt_finish+0x7f/0x150 [xone_gip 31d51abdd4e12bbf79188657a9e679848a766a16]
[42835.810172]  gip_dispatch_pkt.part.0.isra.0+0x667/0xbe0 [xone_gip 31d51abdd4e12bbf79188657a9e679848a766a16]
[42835.810180]  gip_process_buffer+0x15e/0x440 [xone_gip 31d51abdd4e12bbf79188657a9e679848a766a16]
[42835.810190]  xone_dongle_process_wlan+0x243/0x300 [xone_dongle 80dfcd0824fad629bfc9c4c1a968f4bf34602212]
[42835.810206]  xone_dongle_complete_in+0x2a0/0x350 [xone_dongle 80dfcd0824fad629bfc9c4c1a968f4bf34602212]
[42835.810211]  __usb_hcd_giveback_urb+0x9d/0x120
[42835.810221]  usb_giveback_urb_bh+0xc3/0x150
[42835.810225]  process_one_work+0x190/0x350
[42835.810234]  bh_worker+0x1ac/0x210
[42835.810238]  ? sched_clock_cpu+0xf/0x200
[42835.810249]  tasklet_action+0x10/0x30
[42835.810255]  handle_softirqs+0xe3/0x2a0
[42835.810259]  __irq_exit_rcu+0xcb/0xf0
[42835.810263]  common_interrupt+0x47/0xa0
[42835.810267]  asm_common_interrupt+0x26/0x40
[42835.810273] RIP: 0033:0x55b0c5006a2c
[42835.810337] Code: 48 eb d0 62 f1 fe 48 7f 12 48 8d 52 40 83 e9 20 7f d3 c5 f8 77 5d c3 cc cc cc cc cc cc cc 55 48 89 e5 48 29 fe c4 e2 7d 30 07 <c4> e2 7d 30 0c 37 48 8d 7f 10 c5 f5 71 f1 08 c5 f5 eb d0 c5 fe 7f
[42835.810339] RSP: 002b:00007f9c1d5fcd10 EFLAGS: 00000202
[42835.810345] RAX: 0000000000000000 RBX: 00007f9971f2c400 RCX: 0000000000011f80
[42835.810347] RDX: 00007f9971f48100 RSI: 00000000001fa400 RDI: 000030a41965a8c0
[42835.810348] RBP: 00007f9c1d5fcd10 R08: 0000000000000000 R09: 0000000000000f00
[42835.810350] R10: 0000000000000780 R11: 0000000000000f00 R12: 000055b0c5006a20
[42835.810351] R13: 0000000000000001 R14: 000030a419846e40 R15: 000030a41964ca40
[42835.810357]  </TASK>
[42835.810374] Kernel Offset: 0x19000000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
```

cobarx, 2025-08-30:
Looks like you have an old version of the driver, the latest is 0.4.3. Try updating and see if that fixes it.

Lawstorant, 2025-08-30:
Please, post issues only if trsted with newest version

closed 2025-08-30 by Lawstorant

Lawstorant, 2025-08-30:
The issue won't persist. For any other, please create new or post in one that's already open

forest-cat, 2025-08-31:
The issue does not occur with the newest version, so yes its not happening anymore