dlundqvist/xone #115: Xbox Wireless Adapter not working on headless Ubuntu Server 25.04

wes1993, 2025-09-01:
### Description
I’m trying to configure the Xbox Wireless Adapter to share it with a Windows machine using [usbip-win2](https://github.com/vadimgrn/usbip-win2).  
The goal is to keep the dongle connected to a Linux server and use the controller on my Windows PC.

---

### Issue
The adapter does not seem to work:
- The LED stays off.  
- Pairing mode does not activate.  

---

### Steps to Reproduce
1. Run `sudo make test`.  
2. Connect the Xbox Wireless Adapter to the Linux server.  
3. Run `echo 1 | sudo tee /sys/bus/usb/drivers/xone-dongle/*/pairing`
---

### Observed Behavior
- The LED on the dongle never lights up.  
- No response when trying to enter pairing mode.  

---

### Expected Behavior
The dongle should power on, and controller should recognize it so that pairing can be initiated.  

---

### Logs
Relevant `dmesg` output after running `sudo make test`:  
```
[    3.458939] systemd[1]: Started multipathd.service - Device-Mapper Multipath Device Controller.
[    3.460603] systemd[1]: Started systemd-journald.service - Journal Service.
[    3.549006] systemd-journald[309]: Received client request to flush runtime journal.
[    4.620909] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    4.621461] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    4.621692] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[    5.180555] loop0: detected capacity change from 0 to 8
[    5.399415] NET: Registered PF_QIPCRTR protocol family
[    5.519460] piix4_smbus 0000:00:01.3: SMBus Host Controller at 0x700, revision 0
[    5.519636] i2c i2c-0: Memory type 0x07 not supported yet, not instantiating SPD
[    5.578275] xone_gip: loading out-of-tree module taints kernel.
[    5.578285] xone_gip: module verification failed: signature and/or required key missing - tainting kernel
[    5.897366] usb 2-1: reset high-speed USB device number 2 using xhci_hcd
[    5.900793] RAPL PMU: API unit is 2^-32 Joules, 0 fixed counters, 10737418240 ms ovfl timer
[    6.027471] usbcore: registered new interface driver xone-dongle
[  149.081988] usbcore: deregistering interface driver xone-dongle
[  349.069212] usb 2-1: USB disconnect, device number 2
[  361.738930] usb 2-1: new high-speed USB device number 3 using xhci_hcd
[  361.865397] usb 2-1: New USB device found, idVendor=045e, idProduct=02e6, bcdDevice= 1.00
[  361.865408] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  361.865411] usb 2-1: Product: XBOX ACC
[  361.865413] usb 2-1: Manufacturer: Microsoft Inc.
[  361.865415] usb 2-1: SerialNumber: 071161
[  361.972636] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  361.973147] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  361.973377] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[  362.255338] usb 2-1: reset high-speed USB device number 3 using xhci_hcd
[  362.383183] usbcore: registered new interface driver xone-dongle
[  518.464449] usbcore: deregistering interface driver xone-dongle
[  518.527407] usbcore: registered new interface driver xone-wired
[  518.534759] xone_dongle_pre_reset
[  518.806696] usb 2-1: reset high-speed USB device number 3 using xhci_hcd
[  518.934123] xone_dongle_post_reset
[  518.934196] xone-dongle 2-1:1.0: xone_dongle_fw_requester: trying to load firmware xow_dongle_045e_02e6.bin
[  518.934206] xone-dongle 2-1:1.0: xone_dongle_fw_requester: attempt: 1
[  518.935163] xone-dongle 2-1:1.0: xone_dongle_fw_load: firmware requested successfully
[  518.935533] usbcore: registered new interface driver xone-dongle
[  518.936706] xone-dongle 2-1:1.0: xone_mt76_load_firmware: resetting firmware...
[  518.952341] xone-dongle 2-1:1.0: xone_mt76_init_radio: id=0x7612
[  518.988754] xone_gip_headset: Unknown symbol snd_pcm_new (err -2)
[  518.988782] xone_gip_headset: Unknown symbol snd_card_register (err -2)
[  518.989362] xone_gip_headset: Unknown symbol snd_card_new (err -2)
[  518.989380] xone_gip_headset: Unknown symbol snd_pcm_lib_ioctl (err -2)
[  518.989397] xone_gip_headset: Unknown symbol snd_pcm_stream_unlock_irqrestore (err -2)
[  518.989407] xone_gip_headset: Unknown symbol snd_card_free_when_closed (err -2)
[  518.989416] xone_gip_headset: Unknown symbol snd_pcm_set_ops (err -2)
[  518.989445] xone_gip_headset: Unknown symbol snd_card_disconnect (err -2)
[  518.989504] xone_gip_headset: Unknown symbol _snd_pcm_stream_lock_irqsave (err -2)
[  518.989514] xone_gip_headset: Unknown symbol snd_pcm_period_elapsed (err -2)
[  518.991639] xone-dongle 2-1:1.0: xone_mt76_init_address: fuse_address=62:45:b4:ee:55:8c
[  519.001093] xone-dongle 2-1:1.0: xone_mt76_evaluate_channels: channel=1, power=31
[  519.004192] xone-dongle 2-1:1.0: xone_mt76_evaluate_channels: channel=6, power=32
[  519.006380] xone-dongle 2-1:1.0: xone_mt76_evaluate_channels: channel=11, power=32
[  519.009929] xone-dongle 2-1:1.0: xone_mt76_evaluate_channels: channel=36, power=43
[  519.012253] xone-dongle 2-1:1.0: xone_mt76_evaluate_channels: channel=40, power=43
[  519.015338] xone-dongle 2-1:1.0: xone_mt76_evaluate_channels: channel=44, power=39
[  519.017893] xone-dongle 2-1:1.0: xone_mt76_evaluate_channels: channel=48, power=39
[  519.021288] xone-dongle 2-1:1.0: xone_mt76_evaluate_channels: channel=149, power=34
[  519.025148] xone-dongle 2-1:1.0: xone_mt76_evaluate_channels: channel=153, power=34
[  519.027369] xone-dongle 2-1:1.0: xone_mt76_evaluate_channels: channel=157, power=33
[  519.032129] xone-dongle 2-1:1.0: xone_mt76_evaluate_channels: channel=161, power=33
[  519.034149] xone-dongle 2-1:1.0: xone_mt76_evaluate_channels: channel=165, power=33
[  519.034501] xone-dongle 2-1:1.0: xone_mt76_init_channels: channel=36
[  599.476093] xone-dongle 2-1:1.0: xone_dongle_toggle_pairing: enabled=1
```
---

### Environment
- **Linux server**: Ubuntu Server 25.04
- **Kernel version**: 6.14.0-29-generic

wes1993, 2025-09-01:
Yes correct my dongle is this: 

[image]

Lawstorant, 2025-09-01:
On the other topic, what's the use case for that kind of janky setup? The latency alone would probably make it questionable.

wes1993, 2025-09-01:
My setup is as follows:
The gaming PC is in the study, connected to a standard 24-inch monitor.
I use Moonlight and Sunshine to play on my TV, which is in the dining room together with the Linux server (Ubuntu on Proxmox), where there is no coverage for the Microsoft dongle.
My goal is therefore to connect the dongle to the Linux server located in the same room as the TV and control the gaming PC that is in the study.
The LG TV has Bluetooth, but unfortunately the Elite Series 1 controller does not.

Lawstorant, 2025-09-01:
I wonder if the issues are not coming from the missing alsa driver in the Ubuntu headless. The driver might be quietly failing there. I'd have to check it out.

wes1993, 2025-09-01:
I don't know, the xone-dongle driver seems loaded into kernel

Lawstorant, 2025-09-22:
There's something about input events/usb handling in the headless images that's disabled and this causes the pairing to not work. It's not something that's in our scope

closed 2025-09-22 by Lawstorant