dlundqvist/xone #121: GameSir G7 Pro support

JohnTheCoolingFan, 2025-09-18:
Comes with either wired mode or dongle

The wired mode already works, but the dongle doesn't seem to work.

It connects, gets recognized as an HID device, but using something like sdl2-jstest doesn't see any activity (despite being able to list it)

dongle:
```
[40580.759737] usb 3-2.2: New USB device found, idVendor=3537, idProduct=0575, bcdDevice= 1.39
[40580.759745] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[40580.759748] usb 3-2.2: Product: Xbox 360 Controller for Windows
[40580.759751] usb 3-2.2: Manufacturer: Guangzhou Chicken Run Network Technology Co., Ltd.
[40580.914971] hid-generic 0003:3537:0575.000F: ignoring exceeding usage max
[40580.915358] input: Guangzhou Chicken Run Network Technology Co., Ltd. Xbox 360 Controller for Windows Keyboard as /devices/pci0000:00/0000:00:08.1/0000:2f:00.3/usb3/3-2/3-2.2/3-2.2:1.1/0003:3537:0575.000F/input/input44
[40581.053484] input: Guangzhou Chicken Run Network Technology Co., Ltd. Xbox 360 Controller for Windows Mouse as /devices/pci0000:00/0000:00:08.1/0000:2f:00.3/usb3/3-2/3-2.2/3-2.2:1.1/0003:3537:0575.000F/input/input45
[40581.053740] hid-generic 0003:3537:0575.000F: input,hiddev99,hidraw5: USB HID v1.10 Keyboard [Guangzhou Chicken Run Network Technology Co., Ltd. Xbox 360 Controller for Windows] on usb-0000:2f:00.3-2.2/input1
```

vendor id `0x3537`, product id `0x0575`

and just in case, the wired dmesg is:

```
[38616.529991] usb 3-4: new full-speed USB device number 9 using xhci_hcd
[38616.678654] usb 3-4: unable to read config index 0 descriptor/all
[38616.678660] usb 3-4: can't read configurations, error -71
[38617.909990] usb 3-4: new full-speed USB device number 11 using xhci_hcd
[38618.053590] usb 3-4: New USB device found, idVendor=3537, idProduct=1022, bcdDevice= 1.95
[38618.053596] usb 3-4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[38618.053599] usb 3-4: Product: GameSir-G7 Pro
[38618.097743] input: GameSir-G7 Pro as /devices/pci0000:00/0000:00:08.1/0000:2f:00.3/usb3/3-4/3-4:1.0/0003:3537:1022.0008/input/input27
[38618.097942] hid-generic 0003:3537:1022.0008: input,hidraw5: USB HID v1.10 Gamepad [GameSir-G7 Pro] on usb-0000:2f:00.3-4/input0
[38618.102822] hid-generic 0003:3537:1022.0009: ignoring exceeding usage max
[38618.103204] input: GameSir-G7 Pro Keyboard as /devices/pci0000:00/0000:00:08.1/0000:2f:00.3/usb3/3-4/3-4:1.1/0003:3537:1022.0009/input/input28
[38618.253422] input: GameSir-G7 Pro Mouse as /devices/pci0000:00/0000:00:08.1/0000:2f:00.3/usb3/3-4/3-4:1.1/0003:3537:1022.0009/input/input29
[38618.253586] hid-generic 0003:3537:1022.0009: input,hiddev99,hidraw6: USB HID v1.10 Keyboard [GameSir-G7 Pro] on usb-0000:2f:00.3-4/input1
```

always seems to disconnect with a usb device number `X+2` where X is the id that it first connected with

(product id for wired is `0x1022`)

Lawstorant, 2025-09-18:
The issue is, this dongle is unpopular and we don't know if it's a clone of the official one or if it's a custom job. Usually, these controllers don't work wirelessly on xbox so it's most likely custom.

Do you need special software to use it on windows? If it needs a driver and isn't plug and play, we basically can't support it AND it's support would be out of scope for this project.

I could certainly add it to the dongle list and we could do a simple test if you want

JohnTheCoolingFan, 2025-09-18:
The dongle doesn't require any additional support on windows (according to the manual), although I don't see any mention of using the dongle or bluetooth with xbox.

Al-Infinitum, 2025-09-20:
linux 6.16.8.arch1-1

With xone installed on my system, the Gamesir Cyclone 2 controller connects via wireless dongle (RGB lights activate), but is not recognized as a controller by Steam.

If I uninstall xone on a nearly identical system, the controller connects via the wireless dongle and is recognized by Steam.  I started to post this as a new issue, then looked at open issues and caught this one.

Hopefully the side-by-side comparison of my two nearly identical installs clarifies the issue. I suspect adding the dongle to the list will do the trick.

Specifically, when xone is not installed (post-uninstall and reboot), plugging in the dongle and turning on the controller leads to this conclusion:
[  326.502295] input: Generic X-Box pad as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/input/input52
[  326.502440] usbcore: registered new interface driver xpad

dmesg with xone installed:
[  114.038346] usb 1-3: new full-speed USB device number 7 using xhci_hcd
[  114.407251] usb 1-3: New USB device found, idVendor=3537, idProduct=0575, bcdDevice= 1.19
[  114.407258] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  114.407259] usb 1-3: Product: GameSir-Cyclone 2
[  114.407260] usb 1-3: Manufacturer: Guangzhou Chicken Run Network Technology Co., Ltd.
[  114.429091] input: Guangzhou Chicken Run Network Technology Co., Ltd. GameSir-Cyclone 2 Consumer Control as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:0a:00.0/usb1/1-3/1-3:1.0/0003:3537:0575.0009/input/input33
[  114.479363] input: Guangzhou Chicken Run Network Technology Co., Ltd. GameSir-Cyclone 2 as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:0a:00.0/usb1/1-3/1-3:1.0/0003:3537:0575.0009/input/input34
[  114.479529] hid-generic 0003:3537:0575.0009: input,hiddev98,hidraw6: USB HID v1.11 Device [Guangzhou Chicken Run Network Technology Co., Ltd. GameSir-Cyclone 2] on usb-0000:0a:00.0-3/input0
[  136.569771] usb 1-3: USB disconnect, device number 7
[  137.148311] usb 1-3: new full-speed USB device number 8 using xhci_hcd
[  137.514679] usb 1-3: New USB device found, idVendor=3537, idProduct=100b, bcdDevice= 1.19
[  137.514683] usb 1-3: New USB device strings: Mfr=0, Product=2, SerialNumber=3
[  137.514684] usb 1-3: Product: Xbox 360 Controller for Windows
[  137.514685] usb 1-3: SerialNumber: 00006F64096B22E7
[  137.554539] hid-generic 0003:3537:100B.000A: ignoring exceeding usage max
[  137.554683] input: Xbox 360 Controller for Windows Keyboard as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:0a:00.0/usb1/1-3/1-3:1.1/0003:3537:100B.000A/input/input36
[  137.644397] input: Xbox 360 Controller for Windows Mouse as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:0a:00.0/usb1/1-3/1-3:1.1/0003:3537:100B.000A/input/input37
[  137.644608] hid-generic 0003:3537:100B.000A: input,hiddev98,hidraw6: USB HID v1.10 Keyboard [Xbox 360 Controller for Windows] on usb-0000:0a:00.0-3/input1
[  854.905653] usb 1-3: USB disconnect, device number 8
[  855.486784] usb 1-3: new full-speed USB device number 9 using xhci_hcd
[  855.854798] usb 1-3: New USB device found, idVendor=3537, idProduct=0575, bcdDevice= 1.19
[  855.854804] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  855.854806] usb 1-3: Product: GameSir-Cyclone 2
[  855.854807] usb 1-3: Manufacturer: Guangzhou Chicken Run Network Technology Co., Ltd.
[  855.876618] input: Guangzhou Chicken Run Network Technology Co., Ltd. GameSir-Cyclone 2 Consumer Control as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:0a:00.0/usb1/1-3/1-3:1.0/0003:3537:0575.000B/input/input39
[  855.927429] input: Guangzhou Chicken Run Network Technology Co., Ltd. GameSir-Cyclone 2 as /devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:0a:00.0/usb1/1-3/1-3:1.0/0003:3537:0575.000B/input/input40
[  855.927603] hid-generic 0003:3537:0575.000B: input,hiddev98,hidraw6: USB HID v1.11 Device [Guangzhou Chicken Run Network Technology Co., Ltd. GameSir-Cyclone 2] on usb-0000:0a:00.0-3/input0

dmesg with xone uninstalled:
[  307.043242] usb 3-4: new full-speed USB device number 4 using xhci_hcd
[  307.169669] usb 3-4: New USB device found, idVendor=3537, idProduct=0575, bcdDevice= 1.19
[  307.169676] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  307.169678] usb 3-4: Product: GameSir-Cyclone 2
[  307.169680] usb 3-4: Manufacturer: Guangzhou Chicken Run Network Technology Co., Ltd.
[  307.171616] input: Guangzhou Chicken Run Network Technology Co., Ltd. GameSir-Cyclone 2 Consumer Control as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:3537:0575.0009/input/input46
[  307.222422] input: Guangzhou Chicken Run Network Technology Co., Ltd. GameSir-Cyclone 2 as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:3537:0575.0009/input/input47
[  307.222572] hid-generic 0003:3537:0575.0009: input,hiddev96,hidraw8: USB HID v1.11 Device [Guangzhou Chicken Run Network Technology Co., Ltd. GameSir-Cyclone 2] on usb-0000:00:14.0-4/input0
[  325.185597] usb 3-4: USB disconnect, device number 4
[  325.715337] usb 3-4: new full-speed USB device number 5 using xhci_hcd
[  325.841321] usb 3-4: New USB device found, idVendor=3537, idProduct=100b, bcdDevice= 1.19
[  325.841330] usb 3-4: New USB device strings: Mfr=0, Product=2, SerialNumber=3
[  325.841332] usb 3-4: Product: Xbox 360 Controller for Windows
[  325.841334] usb 3-4: SerialNumber: 00006F64096B22E7
[... cut ...]
[  325.923487] hid-generic 0003:3537:100B.000A: input,hiddev96,hidraw8: USB HID v1.10 Keyboard [Xbox 360 Controller for Windows] on usb-0000:00:14.0-4/input1
[  326.502295] input: Generic X-Box pad as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/input/input52
[  326.502440] usbcore: registered new interface driver xpad

lsusb with the controller on:
Bus 001 Device 010: ID 3537:100b  Xbox 360 Controller for Windows

lsusb with the controller off:
Bus 001 Device 009: ID 3537:0575 Guangzhou Chicken Run Network Technology Co., Ltd. GameSir-Cyclone 2

Lawstorant, 2025-09-21:
You need xpad-noone

forkymcforkface, 2025-09-22:
> > [@JohnTheCoolingFan](https://github.com/JohnTheCoolingFan) have you tried without xone? If it works without xone, and you want to keep using xone for wired mode, you need xone-noone
> 
> Tried that, and it did work. At first when I connected and it recognized only as `0575`, scanning for gamepads (`sdl2-jstest -l`) seems to have output some error into dmesg, but then it reconnected as `1022` and worked!
> 
> Now I just have to find a way to install xpad-noone. It doesn't seem to have an ebuild for gentoo, so I'll probably have to write one myself.

If you write it, I'll add it to the repo.

forkymcforkface, 2025-09-22:
> @forkymcforkface writing it right now, what license does xpad-noone use?

GPL2.0

Lawstorant, 2025-09-22:
@Al-Infinitum don't be! The driver situation IS a bit of a mess really.